tinkerspy/Automaton #1: Odd behaviour with messages to Atm_led

igough, 2015-11-02:
I am seeing some odd behavior with messages to Atm_led not being processed when they should. The following test sketch at the bottom shows the behaviour. It is a simple timer with a 5 second tick at which time it sends a message to a LED to either turn it off or on, alternately. I receive the following output:

Starting...
0 Switching LED1 from state *NONE* to IDLE on trigger *NONE* (1 cycles in 0 ms)
Sending MSG_OFF to led
Sending MSG_ON to led
10002 Switching LED1 from state IDLE to ON on trigger EVT_ON (48243 cycles in 10000 ms)
10006 Switching LED1 from state ON to IDLE on trigger EVT_OFF (1 cycles in 0 ms)
Sending MSG_OFF to led
Sending MSG_ON to led
20002 Switching LED1 from state IDLE to ON on trigger EVT_ON (48243 cycles in 9990 ms)
20006 Switching LED1 from state ON to IDLE on trigger EVT_OFF (1 cycles in 1 ms)

which shows the off message being effectively queued at 5 seconds and then processed right after the ON message at 10 seconds.

Also, there is something odd with the code in Machine::cycle()

```
for ( i = ATM_ON_EXIT + 1; i < state_width; i++ ) { 
            if ( ( read_state( state_table + ( current * state_width ) + i ) != -1 ) && ( i == state_width - 1 || event( i - ATM_ON_EXIT - 1 ) ) ) {
                action( read_state( state_table + ( current * state_width ) + ATM_ON_EXIT ) );
                state( read_state( state_table + ( current * state_width ) + i ) );
                trigger = i - ATM_ON_EXIT - 1;
                return *this;
            }
        }
```
where event() is being passed a state instead of an event.

```
#include <Automaton.h>
#include <Atm_led.h>
#include <Atm_timer.h>

#define UPDATE_TIMER 1
#define TOGGLE_TIMER 2
#define MYLED         13

bool allowUpdate = true;

Atm_led     led;
Atm_timer   toggleTimer;
Factory     factory;

void setup(){
  Serial.begin( 115000 );
  Serial.println( "Starting..." );
  led.begin( MYLED ).state( led.IDLE );
  led.label( "LED1" ).onSwitch( sw, "IDLE\0ON\0START\0BLINK_OFF", "EVT_ON_TIMER\0EVT_OFF_TIMER\0EVT_COUNTER\0EVT_ON\0EVT_OFF\0EVT_BLINK\0ELSE" );
  factory.add( led );
  
  toggleTimer.begin().id( TOGGLE_TIMER ).interval( 5000 ).repeat( ATM_TIMER_OFF ).onTimer( toggleUpdate );
  factory.add( toggleTimer );
}

void loop(){
  factory.cycle();
}

void toggleUpdate( int id ){
  if ( allowUpdate==true ){
    Serial.println( "Sending MSG_OFF to led" );
    led.msgWrite( led.MSG_OFF );
  } else {
    Serial.println( "Sending MSG_ON to led" );
    led.msgWrite( led.MSG_ON );
  }
  allowUpdate = !allowUpdate;
}

void sw( const char label[], const char current[], const char next[], const char trigger[], uint32_t runtime, uint32_t cycles ) {
  Serial.print( millis() );
  Serial.print( " Switching " );
  Serial.print( label );
  Serial.print( " from state " );
  Serial.print( current );
  Serial.print( " to " );
  Serial.print( next );
  Serial.print( " on trigger " );
  Serial.print( trigger );
  Serial.print( " (" );
  Serial.print( cycles );
  Serial.print( " cycles in " );
  Serial.print( runtime );
  Serial.println( " ms)" );
}
```

igough, 2015-11-02:
Forget my comment about a state being passed to event(). I see now that state_width is the max of the number of events.

The problem is probably that in Machine::cycle()
```
if ( ( read_state( state_table + ( current * state_width ) + i ) != -1 ) && ( i == state_width - 1 || event( i - ATM_ON_EXIT - 1 ) ) ) {
```
if the Atm_led is in the IDLE state and it gets sent a MSG_OFF, the read_state() call short circuits and event() is not called, meaning the MSG_OFF does not get removed from the msg queue at that point. Only when the SM goes to the ON state does the msg not get ignored and then subsequently processed.

I am not sure if this was your intent, to queue messages until a state is reached to process that message. I would normally assume that an invalid message for the current state would be dropped.

tinkerspy, 2015-11-03:
The read_state macro just reads the state transition table to see if an event should be processed. If there is one further in the if condition the event functon is called with the proper id. If there is just a -1 value the rest of the condition is short circuited, since there is no event to process.

The messaging cue is really just a bunch of counters that can be written (incremented) or read (decremented) and in doing so behave somewhat like a messaging queue end in some other ways not at all.

Note that if you use an invalid message id you are effectively writing outside the messaging queues address space leading to unpredictable behaviour. Automaton does not catch or prevent this error. That kind of thing is expensive on a microcontroller. We must be frugal with code and memory usage.

tinkerspy, 2015-11-03:
I just tried your sketch. The problem is that you start by sending a MSG_OFF message which is not read since the Atm_led machine is in the IDLE state. That's perhaps not the most forgiving behavior.

Easy to fix with a change to the transition table. A MSG_OFF message in the IDLE state is now processed (by switching to IDLE), a MSG_ON in the ON state is now processed (by switching to ON). So they have no effect but removing the superfluous message from the queue.

I updated the Atm_led.cpp file on github with the fix so you can download that.

igough, 2015-11-03:
Your change fixes the LED on/off case, but it does not fix it for any other cases, or any other state machines, including those written by other people.

Unfortunately, this is a generic problem with the code. If a message M is sent while the state machine is in a state that does not handle it (a -1 state) AND a subsequent message M2 is sent that is lower numbered than M and is valid for the current state, then `cycle()` will process the lower numbered M2 first, causing a state change and then M will get processed in the next cycle on the new state. If M is valid in this new state, then it will get processed. So the starting state is START (blinking) and if M is an MSG_BLINK and M2 is an MSG_OFF, then the LED never gets turned off (well, it does for 1 cycle).

This means there are 2 generic problems:

1. Messages sent while in an invalid state for that message are not discarded immediately.
2. Messages are processed according to their value, lowest to highest, so messages can be processed out-of-order with respect to when they were sent.

Fixing the first issue will go a long way towards helping 2, because currently, invalid messages are saved in `msg_table` (by incrementing that message's counter) until the state machine transitions to a state in which it can be handled. This can create a large window of time in which to accumulate a message which could be handled out-of-order. So by dumping invalid messages within
`msgWrite()`, this would be a generic fix for all state machines, including those that are not written by you. And since messages are usually only sent on state transitions, this minimizing the chance of handling a message out of order (though it is still possible with the current message processing logic in `cycle()`)

The following sketch shows the same issue on the Atm_led state machine, using the blink and off messages as described above. This is just meant to be an example of how the issue is a generic problem and that changing each state machine to close up the holes is just playing whack a mole, since there will always be a state machine that you don't own that exhibits the problem.

```
#include <Automaton.h>
#include <Atm_led.h>
#include <Atm_timer.h>

#define UPDATE_TIMER 1
#define TOGGLE_TIMER 2
#define MYLED         13

bool allowUpdate = true;

Atm_led     led;
Atm_timer   toggleTimer;
Factory     factory;

void setup(){
  Serial.begin( 115200 );
  Serial.println( "Starting..." );
  led.begin( MYLED );
  // Send MSG_BLINK to force the SM to the START state and then issue a subsequent
  // MSG_BLINK in toggleUpdate() to cause the msg to be kept when it should not be
  led.msgWrite( led.MSG_BLINK );
  led.label( "LED1" ).onSwitch( sw, "IDLE\0ON\0START\0BLINK_OFF", "EVT_ON_TIMER\0EVT_OFF_TIMER\0EVT_COUNTER\0EVT_ON\0EVT_OFF\0EVT_BLINK\0ELSE" );

  factory.add( led );
  
  toggleTimer.begin().id( TOGGLE_TIMER ).interval( 5000 ).repeat( ATM_TIMER_OFF ).onTimer( toggleUpdate );
  factory.add( toggleTimer );
}

void loop(){
  factory.cycle();
}

void toggleUpdate( int id ){
  if ( allowUpdate==true ){
    Serial.println( "Sending MSG_BLINK to led" );
    led.msgWrite( led.MSG_BLINK );
  } else {
    Serial.println( "Sending MSG_OFF to led" );
    led.msgWrite( led.MSG_OFF );
  }
  allowUpdate = !allowUpdate;
}

void sw( const char label[], const char current[], const char next[], const char trigger[], uint32_t runtime, uint32_t cycles ) {
  Serial.print( millis() );
  Serial.print( " Switching " );
  Serial.print( label );
  Serial.print( " from state " );
  Serial.print( current );
  Serial.print( " to " );
  Serial.print( next );
  Serial.print( " on trigger " );
  Serial.print( trigger );
  Serial.print( " (" );
  Serial.print( cycles );
  Serial.print( " cycles in " );
  Serial.print( runtime );
  Serial.println( " ms)" );
}
```

tinkerspy, 2015-11-03:
This is actually by design. The messages are queued in case the machine will process them later. Consider another led machine. One that pulses a led (or a relay) in response to a MSG1 message then pauses and goes back to an IDLE state. You could send it three MSG1 messages (in one burst) which would make it flash three times. Regardless of when the messages arrive the result will be three flashes. You could also send several different messages to put a machine from STATE1 into STATE4 or so. That's why I called it a queue it has a memory so that messsages can be stored until they become relevant for a state.

In previous versions I used a different signalling method (called signal) which had its own limitations. A way to make a state machine ignore 'illegal' messages would be to explicitely flush the queue in the state init action code with obj.msgClear(), you'd have to do that per state.

tinkerspy, 2015-11-03:
Another thought: you could chain msgClear() & msgWrite( led.MSG_BLINK ) together to get the behaviour you want.

```c++
led.msgClear().msgWrite( led.MSG_BLINK )
```

closed 2015-11-07 by tinkerspy

tinkerspy, 2016-03-03:
Hi igough,

I fixed the Atm_led, Atm_fade and Atm_timer machines to handle this issue. I ran into it myself and it can be needlessly confusing.

The solution I implemented is an 'autoclear' option to msgQueue() that makes the machine clear its message queue on every state switch automatically.

Rgdz,
Tinkerspy