tinkerspy/Automaton #5: Implement member for "previous state"

rawtaz, 2016-02-29:
In an action one can inspect the `current` and `next` members to know the current and next state (the latter only applicable in ON_EXIT), but there's AFAIK no way to know what the previous state was.

Can we get a `previous` member too, so we can get this information? I'd say it should be available in ON_ENTER, ON_LOOP and ATM_ON_SWITCH as well.

rawtaz, 2016-02-29:
PS: I do have a use case for this. I'd like to make a "transition state", which is effectively the same as a regular state but the machine is just meant to go to it to perform an action and then go back to the previous state (whichever that was). If I can see which the previous state was, I can make the action go back to it from this temporary state without having to store the previous state myself or making one of these transition states per every other state i want to go from and back to.

tinkerspy, 2016-02-29:
Sort of like a function call with a return statement? OK, I put a 'previous' field in the Machine object in, I'll put it on the repository after I tested it.

rawtaz, 2016-02-29:
Yeah more or less, and I could use a regular function but it'll be nicer to have a transition state for it as it's triggered by an event, then the code is more coherent.

rawtaz, 2016-03-01:
Any idea when you will push this and the changes for message auto cleaning?

tinkerspy, 2016-03-02:
I'll try to do both on thursday.

tinkerspy, 2016-03-03:
Hi rawtaz,

I pushed the autoclear and 'previous' features to github.
I added the 'autoclear' argument to Machine::msgQueue() to the documentation.

Rgdz,
Tinkerspy

closed 2016-03-03 by tinkerspy