tinkerspy/Automaton #7: Balance between states and ON_LOOP

rawtaz, 2016-03-05:
I'm pondering what a good balance is between using separate states for small/simple tasks a machine performs during runtime vs just putting those tasks in the ON_LOOP action of an existing state, and would like to hear your opinions.

Assume you have an info Display that shows what status a Fan is in. The Fan can be Off, Low and High, and these are the three texts that the Display show depending on the status of the Fan.

The Display has at least two states, which are OFF and ON. In the ON state it should continuously display the appropriate text reflecting the Fan status. In the OFF state it doesn't do anything.

Now, the question is which of the following you do in the Display machine, to manage the updating of the status. For the sake of discussion, this is with the background that the Fan signals the Display when its status change, so the Display will determine whether or not to update its text by looking at whether it has a FAN_CHANGED message.

  - In the ON_LOOP action of the ON state, check if the FAN_CHANGED message is set, and if so update the text.

  - Add a third state named UPDATE (with ON on the ELSE event so it returns to the ON state after work is complete) and an event named FAN_CHANGED (with the trigger being a check for the FAN_CHANGED message), and update the display text in the ON_ENTER action of the new UPDATE state.

To me it's a matter of what constitutes a state and how pragmatic one should be. The additional state approach may be more correct and easy to follow, but at the same time updating the text on the display is such a basic part of being a display that is turned on, that it might not warrant a separate state and it's instead better to be pragmatic and just put the update work in the loop code of the ON state and not complicate things further.

What do you guys think?

tinkerspy, 2016-03-06:
Hi rawtaz,

I kind of feel that ON_LOOP should mostly be avoided. The idea of doing things event based is to do nothing but check events. Only update things when an event is fired.

But I also agree you should be pragmatic. On a multitasking/multiuser system it may make sense to be economical with resources, but if you microcontroller has nothing else to do and the user interface doesn't become unresponsive there's probably nothing wrong with constantly updating the display.

I worked with an Adafruit LCD display before and updates were really slow, so there it made sense to update as little as possible.

Some IT guru once said that premature optimization is the root of all evil, I think that is not always true, but I wouldn't worry about it too much.

Regardz,
Tinkerspy

rawtaz, 2016-03-06:
Thanks for your input thinkerspy!

Yeah I think I'm leaning towards that too, to make a separate state for the example above.

As it is now I'm using ON_LOOP in three places, but only for the purpose of forwarding messages/event triggers. I have a number of components/machines in my application and some of them signals events that should be handled by more than one recipient. So in this case, instead of injecting a reference to each recipient in each sending machine I'm just injecting a controller in each sending machine. The controller then forwards messages from the sending machine to the appropriate recipients.

This is done in the ON_LOOP of the controller, instead of having to create a single state and corresponding events for each of these messages. As the work done in this ON_LOOP is just forwarding messages as opposed to doing something that has a more direct effect on/for the controller I find this to be a reasonable pragmatic balance.

One could instead solve this with service locating through the factory or by having more of a pub/sub queue-like system, but adding that would be overkill for this situation.

The ON_LOOP code looks like this:

~~~
if(msgRead(M_FAN_CHANGE))
{
    this->fanInfoText->trigger(this->fanInfoText->E_UPDATE);
}
if(msgRead(M_TEMP_CHANGE))
{
    this->tempInfoTextUpper->trigger(this->tempInfoTextUpper->E_UPDATE);
    this->tempInfoTextLower->trigger(this->tempInfoTextLower->E_UPDATE);
    this->tempInfoTextInner->trigger(this->tempInfoTextInner->E_UPDATE);
}

I'm closing this issue but anyone is very welcome to add more comments if they wish, consider this an open discussion!
~~~

closed 2016-03-06 by rawtaz