tinkerspy/Automaton #8: Setting default/start state of a machine

rawtaz, 2016-03-07:
Hi,

Is there a way to define, in the machine itself, which state it should go to from `*NONE*` (i.e. the first state it should enter when it hasn't been in a state before)?

I find myself wanting to go to another state than the first one in the state table, in a few machines I have. But some of them are just instantiated and started, they're never affected/triggered/changed by another party but instead just sit there doing their work by themselves.

It would be nice if there's a way to define this in the machine class itself instead of setting it in a call to state() or similar (which I suppose one could do after the .begin() in the main code).

tinkerspy, 2016-03-07:
The default state is set to zero (implicitely) in the Automaton.h file in the Machine (and Tiny_Machine) objects. You can deviate from that it in your machine's begin() method by setting 'next' like this:

```c++
  counter.set( repeat_count );
  next = 2;
  return *this;
}
```

This way the machine will switch to state 2 (from state -1, which means uninitialized) at the beginning of the first cycle.

rawtaz, 2016-03-07:
Nice. Just to clarify though, the only thing needed is the `next = 2`, the `counter.set()` is just some other code you had in your machine but isn't related to this question?

tinkerspy, 2016-03-07:
Yeah, sorry about that, I just copied a snippet from Atm_led which I used to test if it actually worked that way. Just the 'next=2;' is needed.

Regardz,
Tinkerspy

rawtaz, 2016-03-07:
Cheers :)

closed 2016-03-07 by rawtaz