tinkerspy/Automaton #9: Ability to reset timer

rawtaz, 2016-03-12:
Hi,

I have a use case where I have a timer (millis in this case), and during runtime (while within a state and not going to another state) I'd like to reset this timer such that it won't return true for expired() until the amount of time configured in the timer has passed *since the time of the reset I want to do*. Basically saying for example "please don't expire until after another ten seconds".

It seems that the timer uses state_millis in the Machine class, so I guess the closest I can get is to do `timer.set(myTimeout + millis() - state_millis)` which stores a new timeout value that works for the purpose when `expired()` evaluates it, but is there a cleaner way already present? If not, perhaps we should introduce a reset() on the timer, that does basically the previous?

On a related note I realize that since everything uses millis() which will overflow after somewhere around 48-50 days, a timer value cannot be larger than that and still "work" correctly.

tinkerspy, 2016-03-12:
I thought of adding a reset() method that does what you suggest, but I decided against it. The timers are special in the way that the set value survives a state change. They can be 'reused' in the next state without setting or resetting them again. A reset() like you suggest would change the value in a way that the user probably wouldn't expect.

timer.begin( this, 5000 ); // Creates a 5 second timer, timer.value=5000

After 3000 millis you call reset( 5000 ) on it, timer.value now contains 8000
When the next state starts the timer value would not be 5000 as you would expect, but 8000!

There's nothing wrong with your solution as long as you remember to set() the timer again when you enter a new state where you want to use it. But I think a reset() method that would work as described would be too confusing.

tinkerspy, 2016-03-12:
The millis() rollover is a arduino-given. I can't do much about that. But the problem isn't that big. I use the solution advocated in the stackoverflow thread below (compare period, not timestamps):

http://arduino.stackexchange.com/questions/12587/how-can-i-handle-the-millis-rollover

This should allow Automaton to handle one millis() rollover gracefully. So the only problem that remains is that you can't specify a timer greater than 4,294,967,295 (49.7 days) but that's what you get for using a 32 bit variable. ;-)

For any applications that require longer timers I would suggest using a real time clock and monitoring that from a state machine.

tinkerspy, 2016-03-13:
Having thought about it, my last statement isn't even true. 

You can easily combine a timer with a counter to achieve longer delays even within one state. A timer expiry switches the state back to itself while the counter is decremented in the ON_ENTER handler. That way the state refreshes its timer long before any millis() rollover could occur.

To achieve a full year delay you could combine 85,899,345 (number of ms in a day) as a timer value with 365 as a counter value and the machine would stay in that state for (about) a year. 

Such a machine's begin function could look like this:

```c++
static const state_t state_table[] PROGMEM = {
/*                ON_ENTER    ON_LOOP  ON_EXIT  EVT_START  EVT_TIMER  EVT_COUNTER  ELSE */
/* IDLE      */   ACT_INIT, ATM_SLEEP,      -1,      WAIT,        -1,          -1,   -1, 
/* WAIT      */  ACT_COUNT,        -1,      -1,        -1,      WAIT,      FINISH,   -1, 
/* FINISH    */ ACT_FINISH,        -1,      -1,        -1,        -1,          -1, IDLE, 
};
timer.begin( this, 85899345 );
counter.set( 365 );
```
Taking the counter value up to the maximum 65534 would give you about 179 years. Neither your microcontroller nor its programmer could expect to be running after that timer expires ;-)

tinkerspy, 2016-03-17:
I extended the Atm_timer machine with an interval_seconds() method that can handled periods of up to 136.2 years. Documentation has been updated.

Rgdz,
tinkerspy

closed 2016-03-17 by tinkerspy