tinkerspy/Automaton #10: Problem about the time control of led

kenthli, 2016-03-16:
I want to control the light dark period of 3 led but it seems over flow when the time is long (above 10 hours) just like the code below. 
There were no errors after complied the problem but it just cannot work.

Atm_led led1, led2, led3;
Factory factory;
unsigned long light1=43200000;
unsigned long dark1=43200000;
unsigned long light2=57600000;
unsigned long dark2=28800000;
unsigned long light3=72000000;
unsigned long dark3=14400000;

void setup() { 
  led1.begin( 2 ).blink( light1 ).pause( dark1 ); // Setup 
  led2.begin( 3 ).blink( light2).pause( dark2 ); 
  led3.begin( 4 ).blink( light3 ).pause( dark3 ); 
  factory.add (led1 ).add( led2 ).add( led3 );          // Add to a factory          // Start blinking
  led1.msgWrite( led1.MSG_BLINK );
  led2.msgWrite( led2.MSG_BLINK );
  led3.msgWrite( led3.MSG_BLINK );
}

void loop() {
  factory.cycle();

}

tinkerspy, 2016-03-16:
Hi kenthli,

That's not strange at all, since the data type for the pause argument is 'int' which overflows at 32,767. 

My bad, I'll fix the Atm_led code today and push the changes to Github. The Automaton timer objects use uint32_t integers so Atm_led & Atm_fade should do the same.

Rgdz,
tinkerspy

kenthli, 2016-03-16:
I got it
Thank you very much

tinkerspy, 2016-03-16:
OK, I fixed Atm_led, Atm_fade and Atm_timer in the master branch.
You should now be able to go all the way up to 4294967294 milliseconds (49.7 days).
That's one very slow blink ;-)

Let me know if it works for you.

Thanks & rgdz,
tinkerspy

kenthli, 2016-03-17:
It works now
thanks for your fast response

tinkerspy, 2016-03-17:
You're welcome,

Thanks for your contribution.

Rgdz,
tinkerspy

closed 2016-03-17 by tinkerspy