tinkerspy/Automaton #11: A problem about factory

kenthli, 2016-04-04:
int light1=12000;
int light2=16000;
int light3=20000;
int dark1=12000;
int dark2=8000;
int dark3=4000;

void setup () {
  led1.begin( 2 ).blink( light1 ).pause( dark1 ); // Setup 
  led2.begin( 3 ).blink( light2).pause( dark2 );
  led3.begin( 4 ).blink( light3 ).pause( dark3 ); 
  factory.add (led1 ).add( led2 ).add( led3 );         // Add to a factory          // Start blinking
  led1.msgWrite( led1.MSG_BLINK );
  led2.msgWrite( led2.MSG_BLINK );
  led3.msgWrite( led3.MSG_BLINK );
}

void loop(){
  factory.cycle();
}

In the above program, I contain a function to change the value of all light and dark variables. However, the factory.cycle() still using the initialized to maintain the blinking process.
I want to ask is that a normal situation?
What can I do if I want to change value inside blink and pause due to different action?

tinkerspy, 2016-04-04:
That should work without problems. The following sketch blinks a led at 1 blink per second. I added a timer state machine that slowly increases the blink period from 500 ms to 1 ms. If you run it you can see the led starting to blink faster and faster until it's too fast for your eyes.

```c++
#include <Automaton.h>
#include <Atm_led.h>
#include <Atm_timer.h>

// Blinking led speeds up when a timer fires

short const ledPin = 3;

Atm_led led;
Atm_timer timer;
Factory factory;

void timer_callback( int id, uint16_t cnt ) {
  led.blink( 500 / cnt ).pause( 500 / cnt ); // Update the led state machine
}

void setup() {
  factory.add( led.begin( ledPin ).blink( 500 ).pause( 500 ) ); // Initialize a led state machine
  factory.add( timer.begin().interval_seconds( 2 ).repeat( 500 ).onTimer( timer_callback ) ); // Initialize a timer
  led.trigger( led.EVT_BLINK ); // Start the led blinking
}

void loop() {
  factory.cycle(); // Run the state machines
}
```

If you still have problems, please post a minimally functional sketch that demonstrates the issue. That helps me figure out what's wrong.

kenthli, 2016-04-06:
It work if call the "led.blink(light1).pause(dark1)" again inside the function

I notice that it will no work if I just change the value of light1 

I think it is because I do not understand the library enough

Thanks for your patient.

tinkerspy, 2016-04-06:
You're welcome!

closed 2016-04-06 by tinkerspy