tinkerspy/Automaton #12: Commands not being recognized as expected when Atm_command is attached to a SoftwareSerial object

mpenna, 2016-04-14:
I'm not sure if this is an Automaton issue by itself, but I've noticed that commands that otherwise work OK when in conjunction with the Serial stream are only recognized if an extra ' ' (blank space) is added to the end of the command when attached instead to a SoftwareSerial object instance. Again, not sure whose library is at fault here but I think it should warrant some investigation on either part.

tinkerspy, 2016-04-14:
There were some bugs in the command matching of Atm_command. They were fixed in the deveolpment branch, could you give this version a try and let me know if perhaps it fixes your problem? 

https://github.com/tinkerspy/Automaton/tree/53f9078ec75846be7bd564c74373f57e5c06dd47

Changes were:
- commands are now matched fully
- Both CR & NL are now recognized as an end of line character

mpenna, 2016-04-14:
Hi, thank you for the super speedy reply! For good measure (I should've done this in the first place), before updating the library I went on and tested this same functionality with some other application/terminal connected at the other end of this "pipe". When I reported the problem I was using a C# script inside a Unity test app/game. It must have been some problem on my side then, because now that I've retested this with AGG Software's Serial Port Monitor app the commands are all being correctly recognized. I'll have to further investigate this C# script inside the Unity game to try to find what could be the problem. In perhaps you have any ideas whatsoever, I'd more than appreciate. BTW, this library of yours is absolutely fantastic. Congratulations on this amazing piece of software!

tinkerspy, 2016-04-14:
You're welcome. There can be timing issues with SoftwareSerial, especially with ATtiny microcontrollers that use their built in oscillator. Perhaps changing the baudrate could help?

Thanks for the praise, I've started a testimonials page on the wiki and added yours ;-)

mpenna, 2016-04-14:
Thanks for the tip. For now I'm doing my tests on the Arduino Uno (SparkFun's RedBoard version, that is). I'll certainly take a deeper look into all this and share my findings, if any! (LOL). 

Sure, I'm glad to help and take part of your testimonials page!

tinkerspy, 2016-04-15:
You probably won't have timing issues on an Uno because it uses an external chrystal.

closed 2016-04-15 by tinkerspy