tinkerspy/Automaton #13: Assistance/Tips for RF24Network machine

harmon25, 2016-04-17:
Hi tinkerspy,

Really like what you have done with this library, very cool !!

I am working on a home automation / smarthome project, and I would like to create all the nodes as  Automaton factories. Figure this is a pretty common use case, other people can benefit from your feedback here.

One thing that will be consistent for all nodes is sending and receiving messages over a 2.4GHz wireless network. ([RF24](https://github.com/TMRh20/RF24) + [RF24Network](https://github.com/TMRh20/RF24Network) )

Been playing around with a generic RF24Network machine which can be run within a factory and communicate and be triggered by other machines. [This is what I have so far](https://github.com/harmon25/Atm_RF24Network)

It seems to be working(states changing, events/actions triggering) but not convinced the state table is right. 

1. What do you think? Suggestions for how best to pull this off?
2. I am using ON_LOOP for a function which needs to be called regularly within the loop, should i set it up as a timer? (based on best practices?)
3. Have not yet implemented the SEND action which will take data from a buffer and send it over the network. Should I create a buffer in the machine that can be written to and read by the machine when the SEND event is triggered? Any tips on how best to do this?

Will likely have more questions about this and related topics.
Thanks for any help!

tinkerspy, 2016-04-18:
Hi Harmon,

1.
What is the function of your nodes? Do they, for example, switch a lamp on and off and report on the status of the lamp? In that case I'm thinking of three machines. One for receiving commands via the wireless interface and one that represents the lamp. You could just use an Atm_led machine for the lamp. Perhaps add a button to allow switching the lamp on and off 'locally'.

I'm not familiar with the RF24Network library, but I'm thinking it could work much like the Serial object, in the sense that you can write to it without worrying about timing, but you need some kind of polling to read from it. If it's a Stream object you could even use the Atm_command machine to processs commands. If it isn't you could write something custom in a similar manner. Basically you define an event that checks for available incoming data (like Serial.available) and an action that reads the data into a buffer.

2.
I think ON_LOOP would be the right place to put this.

3.
Doesn't the RF24Network do the buffering for you? Otherwise you could create a machine for it. I think I would add a custom method to the 'SENDER' machine like send() that takes a pointer to a string and its length and make it send the data out. And maybe add a busy() method to check if the machine is done sending.

I'm also thinking that it might be more elegant to declare and initialize the network interface inside the state machine so that the state machine represents the whole network interface. You'd have to adapt it to handle both sending and receiving but that shouldn't be too difficult I think. It's nice to hide the complexities of the network interface that way so you end up with a clean .ino sketch. You could make the resulting Network state machine available for other people as an Automaton (sub)library. I might become one of your users myself, it sounds like a great way to do cheap in house networking.

I'm happy to help if you have any more questions.

harmon25, 2016-04-18:
### 1
The idea is to make it generic for different types of nodes, but yes that is a valid example of one.
 I have a prototype node which is the makings of a "smart" outlet. Just a duplex wall socket wired up to 2 relays, 2 LEDs, 2 push buttons, an [nRF24L01](http://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-HowTo) radio module and controlled by an Arduino Pro Mini.

The LEDs indicate the state of the relays, the pushbuttons allow manual control of the relay state, and the radio allows remotely controlling the relay state. (pretty much your lamp example x2 and instead of a lamp, could be any load the relay can handle.)

My current sketch for this uses 2 led machines, 2 button machines, 2 custom relay machines(minimal custom machine), and the Atm_RF24Network machine(7 machines total). Main problem right now is it uses up 72% of dynamic memory on an UNO (of 2k). Should probably combine some of the machines to reduce the number of machines defined globally...any recommendation on the max number of machines one should aim for in a single sketch/factory? 

Yea am essentially polling the network with the network.available() call in the `EVT_RECV` event which will return 1 if there is a message for the node, much like the `EVT_INPUT` event from the command machine. 

Some other node examples would be things like temp sensors that would ping the temp back at an interval and PIR motion sensors that send if motion is triggered etc.

### 2
 network.update() is used to forward packets that are not for me, but another node that might be routed through me (RF24Network details) it is also used for fragmented packets, and automatic ACKs, so yea ON_LOOP seemed like the right place for that.

### 3
I added a [simple send method](https://github.com/harmon25/Atm_RF24Network/blob/master/Atm_RF24Network.cpp#L58) to the Atm_RF24Network machine 

It will switch the machines state to `SENDING` when called....
I fixed the state table as it was wrong before. Is it OK that `SENDING` is just an empty state?
Can attach a value to the class which would more accurately represent if the radio is 'busy'. And use an event to indicate when it is not busy, correct?

I tried encapsulating the RF24 and RF24Network constructors within the machine, but having trouble with initialization...will give it another shot. 
Any tips on overloading the Machine constructor?

Thanks again for the help, your feedback is very much appreciated!

tinkerspy, 2016-04-18:
Hi Harmon,

1
- Have a look at Automaton Tiny, you should be able to reduce your memory usage by 50% by dropping some features (like debugging). The message queue is also dropped, but I plan to drop that in the next release anyway. I think that trigger() provides 90% of the functionality for a lot less SRAM usage.
- Another option would be to a different Arduino compatible mcu like the teensy 3.2. Lots of RAM and high speed in a small package.

3
That SENDING state seems unnecessary. There's no reason IMO you can't just perform some action in a custom method as long as it doesn't block the execution and returns quickly. No reason to force everything in to states ;-)

OTOH, be careful using the state( new_state ) method. It bypasses the state table logic which should be the place where all your state logic is located. Better to add an EVT_SEND and trigger it with trigger( EVT_SEND ), that way the state table still controls state changes.

You could also check from the outside if your machine is in a certain state as a way of checking if its busy: if ( atm_net.state() == atm_net.SENDING ), but that doesn't really seem necessary since you don't need a SENDING state ;-)

I don't use the Machine constructor, but I use the begin() method for all initialization. I read that in an Arduino style guide somewhere. Just declare the RF24 objects in the .h file and do the other stuff in your Machine's begin() method.

Hope this helps.

harmon25, 2016-04-18:
Very helpful thank you!  :+1: 

##### 1.
- removed the messaging from the RF24Network machine, did seem a bit unneeded
- aiming for compatibility with all 328p > Arduinos, but agree the teensy would make it easier for sure

##### 3.
- removed the sending state as suggested
- removed pinging state, figure external timers would be more flexible...allowing multiple etc
- [RF24 constructor](https://tmrh20.github.io/RF24/classRF24.html) requires the CE & CS SPI pins passed to it, not really sure how i could do that inside the machine class...
  - would be nice to abstract all of it within the machine somehow
  - will try by passing radio by reference rather than the network...

I will upload this smart outlet sketch after i play around with it some more.
Will be a more complex Atm_RF24Network example combined with other machines into a factory.

harmon25, 2016-04-19:
- switched everything to tiny machines, as messages and debugging are not too important, gotta leave some room for actual functionality haha
- managed to get the network stuff tidied up by only requiring radio as reference and constructing network object in initialization list

Here is the [relay node modular sketch](https://github.com/harmon25/Atm_RF24Network/tree/master/examples/relay_node), decided to put a tinyled machine inside the relay machine, cycling it ON_LOOP...seems to be working OK, even have it blinking before state change! Also dynamic memory usage is down from 72% to 48%!

Am having one odd problem though, in the [button callback](https://github.com/harmon25/Atm_RF24Network/blob/master/examples/relay_node/relay_node.ino#L57)
-  i would like to send the state of the relay in a payload, but for some reason, [relay1.state()](https://github.com/harmon25/Atm_RF24Network/blob/master/examples/relay_node/relay_node.ino#L63) is returning nothing, when [relay2.state()](https://github.com/harmon25/Atm_RF24Network/blob/master/examples/relay_node/relay_node.ino#L71) seems to be working as intended; returning an integer which should represent the state
- also regardless if i print the state before or after the call to `relay2.trigger(EVT_TOGGLE)` the state integer seems wrong...with [this state table](https://github.com/harmon25/Atm_RF24Network/blob/master/examples/relay_node/Att_relay.cpp#L4) shouldn't OFF=0 and ON=1 ? Seem to be getting the opposite...

### Other thoughts
- I like the idea of removing the message queue from the normal machines, it would be nice if the factory was generic and both normal machines and tiny machines could operate together within a factory, removing messaging would make that easier....

- Also the pinChange event helper is pretty handy and will probably want to use it for things like PIR sensors, do you think it is worth including that in the tiny machine base class? It is not a big function...

tinkerspy, 2016-04-19:
Hi Harmon,

You bring up a lot of subjects at the same time. I think you can embed objects with constructor arguments like this:

```cpp
#include <SoftwareSerial.h>

#ifndef Atm_blink_h
#define Atm_blink_h

#include <Automaton.h>

class Atm_blink: public Machine {

  public:
    Atm_blink( void ) : Machine() { class_label = "BLNK"; static SoftwareSerial  mySerial( 10, 11 ); MS = &mySerial; };

    short pin;     
    atm_timer_millis timer;
    
    enum { LED_ON, LED_OFF };
    enum { EVT_TIMER, ELSE };
    enum { ACT_ON, ACT_OFF };
    
    SoftwareSerial * MS;
    Atm_blink & begin( int attached_pin, uint32_t blinkrate );
    Atm_blink & trace( Stream * stream );
    int event( int id ); 
    void action( int id ); 
};

#endif
```

Here I created a SoftwareSerial object within the blink_modular example. I created the object within the constructor (not sure if the static is necessary) and a pointer to a SoftwareSerial object. Then linked the pointer with the object. Anyway, perhaps avoiding constructor arguments in Arduino libraries would be a good idea. This compiles, but I didn't test the functionality.

tinkerspy, 2016-04-19:
Your relay machine contains a led machine. That can be useful and is an interesting concept (you can even run a factory from the ON_LOOP handler). But the following code inside a state machine's action handler breaks the concept:

```cpp
while ( relay_led.cycle().state() ) {};
```

You're cycling the led machine here, but all other machines are stalled at this point. You should never wait inside a state machine. If you want to make a machine (A) wait for another (B) to reach a certain state you should create an event inside A that checks the state of B, like this:

```cpp
    switch ( id ) {
        case EVT_LED_FINISHED :
            return ( led.state() == led.IDLE );
```
That way all machines keep running at full speed. The cycle().state() construct is really meant for special cases and not inside a machine.

But, frankly, in this case I think I would create a relay machine that controls both the relay and led pins directly. That would make for a much cleaner en efficient solution. As you see every state machine istance carries its own overhead. Your functional unit would be a relay with an indicator led, nice and reusable.

As far as the wrong states returned by state() I have no explanation. That shouldn't happen.

I actually dropped the pinChange() method from the latest development branch, it uses 4 bytes SRAM for every Machine instance and as you found out SRAM is precious on an MCU. But I think I'll change it into a class similar to the timers, that way it will only take up memory if you use itand it could be used inside a Tiny machine as well.

If you have a problem that you would like me to look at it would help if you create a minimal sketch that demonstrates the problem. I think the modular nature of Automaton fits your nodes very nicely, interesting project! 

I see you solved the att_net(radio) problem. That's very nice too.

Rgdz,
Tinkerspy

harmon25, 2016-04-20:
Thanks Tinkerspy!
 
I figured my blinking LED solution was a bit of a hack... going to give it another shot with your suggestion and build a proper state machine to handle the led and relay as a functional unit.

Pretty sure the wrong state issue is related to the hacky machine with an led cycling in it, hopefully a better designed machine will solve that. 

Good idea to create a standalone pinChange class, best of both worlds for those who do not require pinChange in a machine.

Will keep you updated on progress and troubles I run into!

Thanks again, you're brilliant!

closed 2016-04-29 by tinkerspy