tinkerspy/Automaton #14: state() does not always return within Atm_btn machine onPress callback

harmon25, 2016-04-20:
Hi Tinkerspy,

Got my relay node with blinking indicator LED working pretty well, but still encountering this weird bug...

Seem to have the same problem on TinyMachines and Machines.

Created a sketch without any of my custom machines to test it out, running on an Arduino UNO. 

The toggles work, machine seems to be functioning, but that one state() call within the button callback doesn't return anything.

Any ideas what could be causing this behaviour? 

``` c++
#include <Automaton.h>
#include <Atm_button.h>
#include <Atm_led.h>

Atm_led led1, led2;
Atm_button btn1, btn2; 
Factory factory; 

void btn_change( int press, int idx ) 
{
  if ( press ) {
    if(idx == 1)
    {
      // this doesnt work
      Serial.println(led1.state());
      led1.toggle( led1.IDLE, led1.START );
    }

    if(idx == 2)
    {
      // this does??
      Serial.println(led2.state());
      led2.toggle( led2.IDLE, led2.START );
    }
  }
}

void setup() {
  Serial.begin(9600);
  // does not matter how the onPress is defined..
  btn1.begin( 1 ).onPress(btn_change, 1);
  btn2.begin( 2 ).onPress(btn_change, 2);
  led1.begin( 5 ).blink( 20 ).state( led1.START );
  led2.begin( 6 ).blink( 10 ).state( led2.START );
  // works here
  Serial.println(led1.state());
  Serial.println(led2.state());

  factory.add(btn1).add(btn2).add(led1).add(led2);
}

void loop() {
  factory.cycle();
}
```

Thanks again, hope we can figure it out!

tinkerspy, 2016-04-20:
I'm at work so I can't really get into this too deep.

Two observations:
- Preferably don't use state() to directly set a machine'ss state, use  trigger() instead:

```cpp
  led1.begin( 5 ).blink( 20 ).state( led1.START );
  led2.begin( 6 ).blink( 10 ).state( led2.START );

// use instead:

  led1.begin( 5 ).blink( 20 ).trigger( led1.EVT_BLINK );
  led2.begin( 6 ).blink( 10 ).trigger( led2.EVT_BLINK );
```

Similarly toggle a led like this (I will drop the toggle() method in the next version):

```cpp
led1.toggle( led1.IDLE, led1.START );

// use instead:

led1.trigger( led1.state() ? led1.EVT_OFF, led1.EVT_START );
```
Also pin 1 on an Arduino is usually reserved for the hardware serial port, try pin 3 or 4 instead.

Let me know if any of this helped.

Rgdz,
Tinkerspy

tinkerspy, 2016-04-20:
This works fine for me. I simulated a keypress with the autoPress() method, changed the button pins from 1 & 2 to 7 & 8 and changed the serial msg somewhat.

```cpp
#include <Automaton.h>
#include <Atm_button.h>
#include <Atm_led.h>

Atm_led led1, led2;
Atm_button btn1, btn2; 
Factory factory; 

void btn_change( int press, int idx ) 
{
  if ( press ) {
    if(idx == 1)
    {
      // this doesnt work
      Serial.print( "btn1 press, led1 state=" );      
      Serial.println(led1.state());
      led1.toggle( led1.IDLE, led1.START );
    }

    if(idx == 2)
    {
      // this does??
      Serial.print( "btn2 press, led2 state=" );      
      Serial.println(led2.state());
      led2.toggle( led2.IDLE, led2.START );
    }
  }
}

void setup() {
  Serial.begin(9600);
  // does not matter how the onPress is defined..
  btn1.begin( 7 ).onPress(btn_change, 1).autoPress( 1000, 1 );
  btn2.begin( 8 ).onPress(btn_change, 2).autoPress( 1000, 1 );
  led1.begin( 5 ).blink( 20 ).state( led1.START );;
  led2.begin( 6 ).blink( 10 ).state( led2.START );;
  // works here
  Serial.println(led1.state());
  Serial.println(led2.state());
  
  factory.add(btn1).add(btn2).add(led1).add(led2);
}

void loop() {
  factory.cycle();
}
```

Output is what I would expect:

```
-1
-1
btn1 press, led1 state=3
btn2 press, led2 state=3
btn2 press, led2 state=0
btn1 press, led1 state=0
btn2 press, led2 state=3
btn1 press, led1 state=3
btn1 press, led1 state=0
btn2 press, led2 state=0
btn1 press, led1 state=3
btn2 press, led2 state=3
btn2 press, led2 state=0
btn1 press, led1 state=0
btn2 press, led2 state=3
btn1 press, led1 state=3
btn2 press, led2 state=0
btn1 press, led1 state=0
```

I'm also using an Arduino Uno so you should expect to see the same.

harmon25, 2016-04-20:
Awesome thanks!
It was likely related to the pin being reserved for serial, switching button pins resolved `state()` returning nothing for a button on pin 1.

Good idea to remove `toggle()` if it is not best practice to use. I created this example based off of the button example sketch which still includes the `toggle()` method used to toggle an led from blinking to off.

You suggested the following which does not appear to trigger the led from blinking to off, will just keep blinking regardless of the current state after a button press.

`led1.trigger( led1.state() ? led1.EVT_OFF : led1.EVT_BLINK );` 

Here is the fixed sketch, did manage to implement the above through manipulation of the LED machines repeat counter. Is this the right way to pull that off?

Might want to adjust to button example to demonstrate this or similar instead of `toggle()` 

```c++
#include <Automaton.h>
#include <Atm_button.h>
#include <Atm_led.h>

Atm_led led1, led2;
Atm_button btn1, btn2; 
Factory factory; 

void btn_change( int press, int idx ) 
{
  if ( press ) {
    if(idx == 1)
    {
      Serial.print( "btn1 press, led1 state=" );      
      Serial.println(led1.state());
      if(led1.state())
      {
        // set blinking to stop
        led1.repeat(0);
      } else {
        //set led to blink
        led1.repeat(ATM_COUNTER_OFF);
      }
      // trigger blink on and off
      led1.trigger( led2.EVT_BLINK );
    }

    if(idx == 2)
    {
      Serial.print( "btn2 press, led2 state=" );      
      Serial.println(led2.state());
      if(led2.state())
      {
        // stop blinking
        led2.repeat(0); 
      } else {
        // start blinking
        led2.repeat(ATM_COUNTER_OFF); 
      }
      // trigger blink on and off
      led2.trigger( led2.EVT_BLINK );
    }
  }
}

void setup() {
  Serial.begin(9600);
  btn1.begin( 3 ).onPress(btn_change, 1);
  btn2.begin( 4 ).onPress(btn_change, 2);
  led1.begin( 7 ).blink( 100 );
  led2.begin( 8 ).blink( 100 );
  // works here
  Serial.println(led1.state());
  Serial.println(led2.state());
  
  factory.add(btn1).add(btn2).add(led1).add(led2);
}

void loop() {
  factory.cycle();
}
```

tinkerspy, 2016-04-20:
You're right, I verified the problem with trigger() in the master. In the development version it worked fine.

The problem was that the Atm_led machine didn't have enough IDLE/START state in the EVT_BLINK and EVT_OFF columns. An event can only be triggered if there's a value in the corresponding location in the table. That's a feature, but the machine must be built for it ;-)

I pushed the fixed Atm_blink.cpp to the master branch.

Good that you fixed the problem by changing the button pins!

tinkerspy, 2016-04-20:
> Might want to adjust to button example to demonstrate this or similar instead of toggle()

Good idea, I'll do that.

closed 2016-04-29 by tinkerspy