tinkerspy/Automaton #15: Big updates

tinkerspy, 2016-05-25:
I merged the development branch to master today. There are a lot of changes in Automaton.

leseigneur, 2016-05-31:
Hi. I have just discovered your work and am evaluating your library to see if I can use it for my projects. So far I like it a lot, thank you.
There is one specific point I need some help with. I see that you have reworked your code to favor direct triggering. My question :  I have the need for message queue : a machine reacts to triggers by generating commands, those commands are sent into a second machine message queue. The second machine then interprets and run the commands at its own pace.
I don't see how this behavior can be implemented with just triggers.
I still see some msg_queue here and there in the code, can I use that for my async needs ?
Thanks (sorry for my english, not my mother tongue)

tinkerspy, 2016-05-31:
Hi Leseigneur,

Yes. After long soul searching I dropped the msg queue. Nobody seemed to be using the queueing behavior and t confused a lot of people when messages kept hanging in the queue. (see some of the other issues)

My advice would be to add msgWrite()/msgRead() method to your own machine(s) and add a data structure to store the data in the mean time. An external process calls msgWrite() which stores the msg and the event handler reads the message when it's ready to process it.

You could look at my old code (in old commits), you can probably reuse some of that. I think the messaging queue could be implemented as a separate class that one could elect to use in a machine when needed. It's not necessary to have it in the Machine base class where it takes up memory even if one doesn't use it.

Rgdz,
Tinkerspy

leseigneur, 2016-05-31:
Thank you for your answer. I like the idea of using an external message queue and mix it where I need it within a dedicated machine.

tinkerspy, 2016-05-31:
I just thought of another thing: I made trigger() a virtual method so you can override it and make it use the messaging queue for storing the triggers. That way you can still use the standard onChange() etc. connectors to trigger your (queueing) machines.

Rgdz,
Tinkerspy

leseigneur, 2016-06-04:
Hello Tinkerspy,
I'm struggling with trigger() when used from within an action method. 
The trigger is ignored and the state won't change.

What I'm trying to do is to fire a state change from within an action 
method, on the machine itself.
Any idea ?

-- 
jy :)

leseigneur, 2016-06-04:
Hello again,
I may have found a mistake in the example code concerning traces.

In the machine building tutorial, at the bottom of the page you show 
this code :

Atm_blink &Atm_blink::trace( Stream & stream ) {
   Machine::setTrace( &stream, atm_serial_debug::trace,
     "BLINK\0EVT_TIMER\0EVT_ON\0EVT_OFF\0ELSE\0IDLE\0LED_ON\0LED_OFF"  );
   return  *this;
}

I have found that it works fine with one class, but won't randomly when 
another class has traces enabled too.
By "not working" I mean that the wrong symbols will be displayed for one 
of the two classes.

After some research I have found that if I enable traces this way :

Atm_blink &Atm_blink::trace( Stream & stream ) {
static char ts[] ="BLINK\0EVT_TIMER\0EVT_ON\0EVT_OFF\0ELSE\0IDLE\0LED_ON\0LED_OFF";
  Machine::setTrace( &stream, atm_serial_debug::trace,ts);
     return  *this;
}

then it works all the time, the right symbol is always displayed !

It is probably due to the fact that in the top version; nothing prevents 
the compiler from resuing the string memory since it is not declared static.
Hope this helps.

-- 
jy :)

tinkerspy, 2016-06-05:
Hi Leseigneur,

No, trigger is not meant for use within the machine itself, it's even blocked from cycling to prevent recursion. You can use the state( new_state) method for that (that form is not available outside the machine).

Do you have a minimal example that shows the tracing problem you describe? (preferably with the bundled machines) I've regularly used tracing on multiple machine types and I never saw that problem myself.

In other news:

I changed the framework so that you no longer need to use the app.component() calls to register a machine. Check the documentation for examples.

I have a prototype for a Machine Editor ( http://wolkendek.nl/atm/ ) that helps in designing the state table for a State Machine and generates skeleton code for the .hpp and .cpp files. I'd like to hear your opinion on that.

Rgdz, Tinkerspy

leseigneur, 2016-06-05:
Hello Tinkerspy,
I can send you my code, but it won't be minimal. As soon as I try to 
reduce it, the problem vanishes. Can I send you the whole thing (I will 
need a mail address please) ?
I have already written a readme word document that describes the steps 
to reproduce it.
All you need is a Arduino Mega (I need to shrink all my arrays to make 
it fit on a Uno, making the problem disapear.) with no additionnal 
hardware, then send a command from serial monitor.

I have tried to import a sdml file in the Machine editor but nothing 
happens when I click import. I have taken my sdml from there 
http://smdl.sourceforge.net/doc/Manual.html#The+SMDL

Thanks for the state(nextState) method, I have overlooked it !

jy :)

tinkerspy, 2016-06-05:
Hmm, you can send me the whole thing if you want but I only have an UNO, a teensy 2.0 and a teensy 3.2. Mail is tinkerspy@myown.mailcan.com.

As far as the editor is concerned, SDML is my own design XML format and has not connection to the http://smdl.sourceforge.net thing, I guess I'll have to switch extensions. Perhaps I'll go with AML, unfortunately almost everything that ends with ML is taken ;-)

If you want to try the editor:
- Enter the name of your state machine in the text box
- Press 'New machine'
- Click on the machine name

The editor will open and you can add states and events en link them together. When you're satisfied click on the .cpp and .h tabs for the generated C++ code.

You can export your machine(s) to a 'SMDL' file (my idea of SMDL) and later import them to continue editing.

Rgdz, Tinkerspy

leseigneur, 2016-06-05:
> tinkerspy@myown.mailcan.com <mailto:tinkerspy@myown.mailcan.com>.

I'll dig into this memory issue, I smell I am responsible for this & you 
won't be able to reproduce on UNO. I send the code to you anyway, if you 
want to have a look.

I love the editor. Easy to use. I WILL use it for my other machines 
since I am only at the beginning of using your library.
Next step would be a lot of work for you: ability to modify an existing 
machine WITH custom code in it. Frankly it is already very helpfull as is.

Using the editor brings two questions :
- how do we implement the EXT_STILL event ?
- I see that I can select a state in the ELSE pseudo Event, does that 
mean that if no event has fired, ELSE it taken in account ?

thanks
jy

tinkerspy, 2016-06-06:
> I'll dig into this memory issue, I smell I am responsible for this & you won't be able to reproduce on UNO. I send the code to you anyway, if you want to have a look.

OK

> I love the editor. Easy to use. I WILL use it for my other machines since I am only at the beginning of using your library.

I use it myself, it's much more convenient than typing al those tables by hand ;-)

> Next step would be a lot of work for you: ability to modify an existing machine WITH custom code in it.

That would be a nice feature, agreed. As you can see the editor already stores the SMDL (ATML?) data inside the .h file. With that info the original template could be recreated and then all matching lines in the custom coded files could be replaced with equivalent from the new template. Something like that...

> how do we implement the EXT_STILL event ?

I don't understand. What's the EXT_STILL event?

> I see that I can select a state in the ELSE pseudo Event, does that mean that if no event has fired, ELSE it taken in account ?

Yes, a row of events in the state table could be read as:

if ( event_1 ) then state_x
if ( event_2 ) then state_y
else state_z

If no other event is triggered, then the machine will switch to the state in the ELSE column, if one is present. The ELSE column also has a handy function as an end-of-row marker.

leseigneur, 2016-06-06:
The EXT_STILL is generated by your editor when I check the Still box on an
event row

tinkerspy, 2016-06-06:
There is no still box unless you create an event or state named that way. Otherwise send me the SDML and I'll check it out.

I thought about the template w/modifications issue and I think it's really just a versioning issue. The new template (B) is a branch off the original template (A), your modified version (C) is another branch. Now it's just a question of merging them. There's no reason git couldn't do that...

leseigneur, 2016-06-06:
Oups, sorry, I meant "Sleep" checkbox in a purple column, not "Still".
I agree it is not necessary to spend time on your editor for Machine 
updates.
Bu the way :
Have you heard about this project : Qp-nano (for Arduino) from 
http://www.state-machine.com/

The machine editor (QM) is very nice, and free, well everything is free. 
The programmer has implemented hierarchical states machines, event 
queues,... the whole 9 yard !

However I have dropped the idea of usine his product in favor of your : 
lighter, faster to learn. I did not want to invest 3 weeks of learning 
just to program my house automation.

Back to the issue I have seen using the setTrace method :
On a C++ phylosophical level, what do you think of the fact that your 
code stores a pointer to a char[] that was defined locally in a method. 
I am not good enough in C++, but I have the feeling that it is not 
correct to assume that the compiler would guaranty access to what is a 
non static local variable outside of the method scope (the method will 
be exited at the time the pointer will be dereferenced).

I ask the question because I have not been able to find an answer by 
myself (again, the memory issue may be mine, but I don't like not being 
able to decide whether the call to setTrace is "C++" valid or not as is).

I guess it is a good question for stackoverflow.com :)

tinkerspy, 2016-06-06:
> I guess it is a good question for stackoverflow.com :)

Right you are. It was already on there (like many, many, other questions I ask myself):

http://stackoverflow.com/questions/9970295/life-time-of-string-literal-in-c

>However, What you have here is an string literal, which is allocated in an implementation defined read only memory. String literals are different from local variables and they remain alive throughout the program lifetime.They have static duration [Ref 1] lifetime. 

It seems to me this shouldn't be a problem.

tinkerspy, 2016-06-11:
@leseigneur 

I just found an error I introduced on july 4th that kind of messed up the tracing. Perhaps that was related to your problem.

Also, I updated the Machine Template Editor: http://wolkendek.nl/atm/
- Better layout
- More intuitive (hopefully)
- Modify existing code (git merge -X ours)
- Support for connectors (onChange etc.)
- Available on github: https://github.com/tinkerspy/Automaton-Machine-Editor

Just import the source code of Atm_player (.hpp & .cpp) if you want to see how it works.

leseigneur, 2016-06-11:
Thank you, I'll try it soon.

I'll tell you if I don't see the weird strings issues with traces.

Do you have some doc on connectors you are willing to share ?

Here is what I see on the cpp page 
"Warning: file_get_contents(machines/stf42cs7o14e8ob0e2inuk0m87/work/Machine.cpp): failed to open stream: No such file or directory in /shared/peter/projects/Editor.prod/cppcode.php on line "

jy :)

tinkerspy, 2016-06-11:
Yeah, fixed that, the 'machines' directory had disappeared.

Basically the connectors (I need to document them on the communication page)  are the on* functions that connect your machine to other machines or callbacks.

```c++
  led.begin( 4 );

  button.begin( 2 )
     .onPress( led, led.EVT_TOGGLE );
```

The onPress() activates up the connector which triggers the led machine.

To create the onPress() method in the Machine editor/Connectors page  just enter 'press' in the text field and press 'Add connector'. Leave the settings at default.

Then in the generated template choose the appropriate place in the action handler to call the connector:

```c++
  case PRESSING:
        push( connectors, ON_FINISH, 0, 1, 0 );

```

That's it.

There are options to create indexed connectors, autostore and broadcast and I really must document them ;-)

closed 2016-07-14 by tinkerspy