tinkerspy/Automaton #19: Atm_Traffic_Light Compile issue

GerryGit, 2016-07-03:
'Atm_trafficlight' does not name a type.
I managed to build the tutorial up to the first compile, which was successful.
The steps thereafter also made sense, but I have run into a compile problem.

I suspect it relates to differences in the generated code and the tutorial as follows: 

Atm_Traffic_Light Traffic_Light;
Atm_trafficlight trafficlight;

This is a bit confusing.

Could you in your follow up please expand on the difference between the upper and lower case instances, as well as publishing a complete code for the traffic light for comparison. 

Very interesting tutorial, thanks.

Gerard

tinkerspy, 2016-07-03:
Hi GerryGit,

As far as I can see the tutorial only mentions 'Atm_trafficlight', if you follow it the generated code should follow the same convention. Perhaps you named your generated machine differently, the tutorial says:

 > Then think of a good name for your machine (Atm_trafficlight would be a good one) and press Rename state machine. 

Perhaps I should have worded it more strongly. ;-)

I'll see if I can find the source I created in the tutorial.

Mike58501, 2016-07-03:
Hi TinkerSpy,

For the sake of possibly gaining some understanding ... I put the -1 "do nothing" in the green, yellow, & red events & then watched my leds.  The state machine resulting behavior seemed to be the same.  I took it a step further because I had been contemplating using the Next event in my washing machine sequence project ... so I put in -1 there.  That did make the auto-sequencing of the leds stop so it appears like the Next event was actually needed & the timer expiring or exit action was not enough to thrust the program into the following row of both action & testing for true or false.  If any one of the events do not test true ... does that mean that the program goes into a wait stage?

My immediate thought was that you are setting the stage for some of the "to be continued" more capability to the Atm_trafficlight full fledged state machine.

Thanks ... Mike

tinkerspy, 2016-07-03:
I added a manual (advance/next) sketch to the tutorial and a half-automatic sketch with two buttons.

Rgdz,
Tinkerspy

tinkerspy, 2016-07-04:
I just added an example to the tutorial that demonstrates how to control the traffic light machine with a web browser by using an Esp8266 wifi-enabled module like the Wemos D1 mini ($4 at Chinese stores).

Rgdz,
Tinkerspy

Mike58501, 2016-07-05:
Hi Tinkerspy,

I was very impressed & can see the potential "in general" value in both of your added button controls & then the wifi (wow) control.  Since I had a momentary button ready to go ... I put its pin # in the parameter & it worked perfectly.  The parameters are setup so the NEXT event is triggered each time the button is pressed which could correlate to an automatically generated momentary digitalRead source which would have the same effect as if the button was physically pushed manually.

Let's say the need arose where the digitalRead had to be about 30 seconds instead of momentary?  I replaced "onPress" with "longPress" & I knew I was in trouble when the color of the keywords did not change & of course it did not compile stating an enum error among other errors.

Can you show us the code changes required within trafficlight "with button" in regards to the connectors, cpp, & h components for the ability to use longPress with the required pin & delay parameters?  Below is your pertinent code that I tried to change the "onPress" with "longPress".  I am thinking you defined the onPress properly but if longPress is used in its place ... it has to also be properly defined & arranged etc ... but how?
~~~
Atm_button button;

void setup() {

  trafficlight.begin( 4, 5, 6 ); // Pins 4, 5 & 6

  button.begin( 17 )
   .onPress( trafficlight, trafficlight.EVT_NEXT );  // longPress applied here technicalities???? 
}
~~~

tinkerspy, 2016-07-05:
You really have to read the documentation, longPress() is not a replacement for onPress(), it modifies the behavior of onPress().

Also read the docs on the digital machine, cause that's the one you need for reading pulses, button/longPress() could work, but digital is better. If you need to detect a 30 second LOW 'pulse', use this instead:

```c++
Atm_digital input;

void setup() {

  trafficlight.begin( 4, 5, 6 ); // Pins 4, 5 & 6

pulse.begin( 17, 30 * 1000, true, true )
    .onChange( trafficlight.EVT_NEXT  );
}
```
Good job with the code display this time!

GerryGit, 2016-07-05:
Thanks Tinkerspy. I have solved all of my compile issues. I learned a little bit about attention to detail. I found it worthwhile to go through Tutorial 2 a few times for proper understanding to sink in.

tinkerspy, 2016-07-05:
Glad to hear it, Gerry,

Automaton has two conceptual levels, the level of the interacting components and the inside of the state machines. When I started building it I was just trying to create a nice way to build state machines, only later I learned that they coud fit together quite elegantly too.

Rgdz,
Tinkerspy

closed 2016-07-14 by tinkerspy