tinkerspy/Automaton #20: Compilation Issues on Attiny85

mannkind, 2016-07-03:
I've been using the Automaton library on an Attiny85 since before the "Big Changes" commit that removed the tiny factory and tiny machines. I recently updated the Automaton library to 121b4dfc61508b3a9f1be115e9cf4a47cc3995d8 (currently the latest commit) and the one sketch I have for an Attiny85 started failing with the following error.

```
.pioenvs/trinket5/Automaton/Atm_servo.hpp:4:19: fatal error: Servo.h: No such file or directory
#include <Servo.h>
```

I'm not using anything Servo related in my sketch. This is the extent to which I use the Automaton library.
```
Appliance app;
Atm_button lampOnBtn;
Atm_button lampDemoBtn;
Atm_led mp3PlayBtn;
Atm_led mp3NextBtn;
Atm_led mp3PrevBtn;
```

After trying a number of previous commits, I found that 0171a9878f5236f39d29f6a383a34937cf9c8169 was the last one where it compiled without issue. As soon as I try to use 0ed4594308b7e98572d7bdf5f353fe8db5ae0b46 or beyond, compilation starts failing with the following error.

```
.pioenvs/trinket5/Servo/avr/Servo.cpp: In function 'void __vector_3()':
.pioenvs/trinket5/Servo/avr/Servo.cpp:82:44: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint16_t* {aka volatile unsigned int*}' for argument '2' to 'void handle_interrupts(timer16_Sequence_t, volatile uint16_t*, volatile uint16_t*)'
handle_interrupts(_timer1, &TCNT1, &OCR1A);
^
.pioenvs/trinket5/Servo/avr/Servo.cpp: In function 'void initISR(timer16_Sequence_t)':
.pioenvs/trinket5/Servo/avr/Servo.cpp:129:5: error: 'TCCR1B' was not declared in this scope
TCCR1B = _BV(CS11);     // set prescaler of 8
^
.pioenvs/trinket5/Servo/avr/Servo.cpp:136:5: error: 'TIFR1' was not declared in this scope
TIFR1 |= _BV(OCF1A);     // clear any pending interrupts;
^
.pioenvs/trinket5/Servo/avr/Servo.cpp:137:5: error: 'TIMSK1' was not declared in this scope
TIMSK1 |=  _BV(OCIE1A) ; // enable the output compare interrupt
^
icons: *** [.pioenvs/trinket5/Servo/avr/Servo.o] Error 1
```

I can certainly use 0171a9878f5236f39d29f6a383a34937cf9c8169 for this specific sketch, but is there anything else I can do on my end? Or is there something Automaton can do?

tinkerspy, 2016-07-03:
I actually moved the Atm_servo machine to its own library 3 days ago. It's better to keep it apart because it introduces a dependency on a third party library (as you just noticed) and I also discovered that it uses 32 bytes of SRAM even when not used. (especially annoying on an ATTiny)

Anyway, if you use the most recent commit both your problems should disappear.

mannkind, 2016-07-03:
Thank you for the fast response! 

I see the [commit](707311790227f91a0e56a0109e79211f85b5b374) that mentions moving it out of this repo, but I still see the [Atm_servo.hpp](https://github.com/tinkerspy/Automaton/blob/master/src/Atm_servo.hpp) and [Atm_servo.cpp](https://github.com/tinkerspy/Automaton/blob/master/src/Atm_servo.cpp) in the latest commit. I feel like I'm missing something simple… haha.

tinkerspy, 2016-07-03:
No, that was just me. It usually takes me a few days after removing something to figure out I shouldn't just delete the files but run 'git rm' on them too. It took me a bit longer this time :-(

(still can't figure out why git doesn't do that automatically)

mannkind, 2016-07-03:
Ah-ha! It's all good now. 

Thanks again for the assistance and awesome library. :)

closed 2016-07-03 by mannkind