tinkerspy/Automaton #21: Atm_button AUTO conflicts with another library

mannkind, 2016-07-03:
Just a note that [Atm_button's AUTO](https://github.com/tinkerspy/Automaton/blob/master/src/Atm_button.hpp#L7) conflicts with the very popular [MySensors](https://github.com/mysensors/Arduino/blob/be92707cc686be14c95686363dd78f3550eda284/libraries/MySensors/core/MyTransport.h#L29) library. 

I end up manually changing Atm_button's AUTO to something else in order to compile my sketches. Which of course, I end up doing each time I update Automaton… :) Maybe there is a better solution on my end, but if so, I'm not familiar with it.

Thank you for your work! Even though I only use a small part of Automaton, it has made my code so much more concise and understandable!

tinkerspy, 2016-07-03:
> I end up manually changing Atm_button's AUTO to something else in order to compile my sketches. Which of course, I end up doing each time I update Automaton… :) Maybe there is a better solution on my end, but if so, I'm not familiar with it.

It's nicely packed inside its own namespace (class) so it must be the other library that's misbehaving :-( Still, it's easy to fix since it's only used internally. I changed it in the last commit.

> Thank you for your work! Even though I only use a small part of Automaton, it has made my code so much more concise and understandable!

Thanks, that's a great compliment!

mannkind, 2016-07-03:
Thanks for changing it despite another library being the one that is misbehaving. :)

closed 2016-07-03 by mannkind