tinkerspy/Automaton #23: Transition to States improved function NEED

Mike58501, 2016-07-24:
[SimplifiedTransitions.zip](https://github.com/tinkerspy/Automaton/files/380072/SimplifiedTransitions.zip)

Hi Tinkerspy & hopefully others,

Anyone confused by this post can get clarification at Tinkerspys YouTube at:
https://www.youtube.com/watch?v=vKqEeH4OvqU

Directly below is a quote from Automaton's Introduction **Mix & match**   

"You can combine Automaton's event driven code with regular Arduino code as long as you make sure that the automaton.run() method is regularly called to update the running state machines."

Below is hopefully a clear enough explanation of why brainstorming on proper mixing could be very helpful & complement the Automaton Framework to be more versatile & useful.

Attached is a simplified (scaled down to bare necessities) zip folder of a sketch with the needed custom machine that compiles but appears to need a "regular Arduino multiple if statement with results going to State rows in a custom machine" made as a key function to be run as the coin credit total current inserted amounts reach the different thresholds.  That in turn goes to one of four holding states that waits for a relatively simple momentary button press (digital input) OV to an active low pin.  Then one of the 4 different relatively simple sequences can proceed similar to the TrafficLight example.  It will also along with a way to have a countdown timer display & be updated (once or twice if not too hard) with a new starting number.  Initially, the countdown & non countdown displays will go to the serial monitor which I should be have no trouble at that time putting it on our 20 x 4 backlit LCD.  Like I said we are not concerned about the time countdown or the  lcd at all at this time.  Like Tinkerspy stated ... First things first!

The "baby step" approach at this stage as recommended by Tinker does compile but I am not seeing the functionality ability in the docs of going from the sketch to changing to the appropriate STATE based on a range of the number of the credit (v) variable & not just by the more limited Automaton Controller Methods.  The Automaton Controller method appears to be locked into the more primitive + - > < etc without the ability to to do the c++ " || (or) etc."   on a somewhat simple ...  albeit temporarily changing credit variable.  A more robust "if function" is needed o prevent non starts of the customer using the coin op accidentally overpays mistaking the higher denomination coin for the lower value coin.
```
 f1_ctl.begin()
  .IF( acceptor, '-', 6 ) //This function is perfect for keeping the green ring led on until
  .led(16 );              // it is turned off one one of the 4 sequences start.

  f2_ctl.begin()
  .IF( acceptor, '=', 8 )  //This is the first of 4 functions that need to allow for
  .led(13 );                  // credit (v) 7 or 8 , 9, 10 or 11. and last function 12, or 13 or 14 or 15
}                             // so if the customer accidently puts in a dollar coin instead of a quarter
// the proper sequence will still start. ALSO INSTEAD WE NEED TO FIND A WAY
void loop() {                 // TO instead of .led we have to have it go to a specific sequence state.
  automaton.run();
```
Thanks Mike

leseigneur, 2016-07-24:
Hello Mike,Here a few advices :- drop the keyboard- take pen and paper, sketch state machines and run in your mind your project.- only when you get confident you have it running smoothly on paper go back to programming- learn how to code. I remember tinkerspy giving you several advices on that subject- if you can't do it yourself,  seek profesional services. After all, this place is about tinkerspy automaton code, not about your.Sorry to be so direct, bit I think you are not in the right direction.Jean-yves

tinkerspy, 2016-07-25:
Thanks Jean-Yves, sorry Mike,

This is not the place for general Arduino help.

closed 2016-07-25 by tinkerspy