tinkerspy/Automaton #29: A question about LED machine

InfinityCode777, 2016-10-18:
The following is the sample code provided in LED machine session. 

`led1.begin( 4 ).blink( 500, 500, 3 ).onFinish( led2, led.EVT_BLINK );`

`led2.begin( 4 ).blink( 50, 50, 10 );`

`led1.trigger( led1.EVT_BLINK );`

Then I have a weird thought that what if I want to repeat the whole process (3xSlowBlink+10xFastBlink) for 10 times or forever? I tried to add the following in the void loop() but it does not work, so any solution to that? Thanks so much :-)

`void loop() {`

  `led1.cycle();`

  `led2.cycle();`

  `//automaton.run();}`

tinkerspy, 2016-10-19:
This is probably your complete sketch:

```c++
#include <Automaton.h>

Atm_led led1, led2;

void setup() {

  led1.begin( 4 ).blink( 500, 500, 3 ).onFinish( led2, led2.EVT_BLINK );

  led2.begin( 4 ).blink( 50, 50, 10 );

  led1.trigger( led1.EVT_BLINK );
}

void loop() {
  automaton.run();
}
~~~

To make it repeat indefinitely you could just add an onFinish() to the second led object like this:

```c++
#include <Automaton.h>

Atm_led led1, led2;

void setup() {

  led1.begin( 4 ).blink( 500, 500, 3 ).onFinish( led2, led2.EVT_BLINK );

  led2.begin( 4 ).blink( 50, 50, 10 ).onFinish( led1, led1.EVT_BLINK );

  led1.trigger( led1.EVT_BLINK );
}

void loop() {
  automaton.run();
}
~~~

Making it repeat a certain number of times is a bit harder. You could trigger the led1 blink sequence with a timer and  make the timer repeat n times.

~~~c++
#include <Automaton.h>

Atm_led led1, led2;
Atm_timer timer;

void setup() {
  
  led1.begin( 4 ).blink( 500, 500, 3 ).onFinish( led2, led2.EVT_BLINK );

  led2.begin( 4 ).blink( 50, 50, 10 );

  led1.trigger( led1.EVT_BLINK );
  
  timer.begin( 5000 )
    .repeat( 3 )
    .onTimer( led1, led1.EVT_BLINK )
    .start();
}

void loop() {
  automaton.run();
}
~~~

If you want real flexibility with led blinking patterns have a look at the [led-controlling example](/tinkerspy/Automaton/wiki/The-player-machine#atm_player--speed-int-v-) in the Player machine documentation. The [sos examples](/tinkerspy/Automaton/wiki#examples) may also be useful.

InfinityCode777, 2016-10-26:
Thanks tinkersky for such a detailed explanation, will play with the code tomorrow and give the feedback.

InfinityCode777, 2016-10-31:
Hi, Tinkerspy, 

Thanks for the previous answer, here is the some feedback. Now, I understand more about the how combine different machines together. All the codes are working example the last example, where the LED will blink according to defined pattern only once the LED is always on. I reduce the pattern array and only keep the first row. I saw the LED on and off for half sec and then LED is always on.

tinkerspy, 2016-11-01:
Do you mean that the Atm_player solution posted here above doesn't work for you? I just copied/pasted it and ran it on my Arduino Uno and it runs perfectly. 4 times the cycle and then off. You didn't wire your led to be active low, perhaps? You can do that too with ```led.begin( 4, 1 );```

InfinityCode777, 2016-11-04:
Hi, Tinkerspy, 

It's my bad and I should have told you earlier that I am using Teensy 3.2 instead of Arduino UNO. Cuz the both platforms are so similar in so many ways and I always believe that codes for two platforms will be 100% compatible. I tried your code on my UNO, it was a success. But on Teensy 3.2 the code runs only ones (That's so weird). Thank you so much for your help.

Also, a last small question, does the number inside "timer.begin(TIME)" denote the duration of the timer, say, timer.begin(5000) means, timer is gonna count 5000 milli seconds, is that right?

Best

Jing

Mike58501, 2016-11-04:
Hi Jing,

On your last small question ... TinkerSpy may add or correct what I say if he is not too busy  ... I am not sure ... as a caveat ... I am still learning myself on the inside workings of the currently existing Automaton State Machines.

Just in case it helps ... the way I read the public declarations in the Atm_timer.hpp & its .cpp file ... the begin method declarations have the beginning interval being set as default 0 ().  By putting in timer.begin(5000) that changed the chosen millisecond interval to that which is = to 5 seconds.  There is a provision for a second arg for the number of repeats so instead of the default repeat being left as 1 it could be changed with the timer.repeat method.  Tinkerspy's above timer example IMHO could have optionally been written timer.begin (5000,3).  In short = the pertinent use of 5 seconds of time repeated 3 times.  Two args not just one.  Some programmers also describe args as arguments or parameters. 

Now to see if I absolutely must include timer states within a larger custom state machine to be able to reactive state machine control " the nontimed condition when" of a timer stop() method & "the nontimed condition when" of a newly defined timer.begin &/or started with a new timer.repeat method or arg.  Anybody up to helping me & possibly others to rise to that LCD. Print Display needed reset loose end ... that I have left in a very important project?

tinkerspy, 2016-11-05:
Hi Jing,

Automaton should work on all Arduino compatible microcontrollers, I try to keep it as portable as possible. I have actually done a lot of my development on Teensy's. I have a teensy 3.1 lying aroung I'll try running the Atm_player code on that, see what the problem is.

What you say about the timer is correct, the first argument is the number of milliseconds for the timer to run, the second (undocumented I see) is the number of repeats (default 1) as Mike points out.

```c++
 
  led.begin( 4 );

  timer.begin( 1000, 10 )
    .onTimer( led. EVT_TOGGLE )
    .start();

```

The code above should make the led blink five times.

InfinityCode777, 2016-11-08:
Hi, @tinkerspy, 

Thanks for the clarification and explanation. Well, I tried that the code with "player.trace( Serial );" again on Teensy 3.2. I got the same results, i.e. the pattern could only be implemented once. Here is what is pulled out into serial monitor:

`900 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (53726 cycles in 500 ms)
1400 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (53175 cycles in 500 ms)
1400 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
1900 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (53651 cycles in 500 ms)
2400 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (53184 cycles in 500 ms)
2400 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
2900 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (53651 cycles in 500 ms)
3400 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (53184 cycles in 500 ms)
3400 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
3450 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5344 cycles in 50 ms)
3500 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5306 cycles in 50 ms)
3500 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
3550 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5344 cycles in 50 ms)
3600 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5307 cycles in 50 ms)
3600 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
3650 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5343 cycles in 50 ms)
3700 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5307 cycles in 50 ms)
3700 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
3750 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5343 cycles in 50 ms)
3800 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5307 cycles in 50 ms)
3800 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
3850 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5344 cycles in 50 ms)
3900 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5306 cycles in 50 ms)
3900 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
3950 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5344 cycles in 50 ms)
4000 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5306 cycles in 50 ms)
4000 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
4050 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5344 cycles in 50 ms)
4100 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5306 cycles in 50 ms)
4100 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
4150 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5344 cycles in 50 ms)
4200 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5306 cycles in 50 ms)
4200 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
4250 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5344 cycles in 50 ms)
4300 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5306 cycles in 50 ms)
4300 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
4350 Switch PLAYER@1FFF9164 from SOUND to QUIET on EVT_TIMER (5344 cycles in 50 ms)
4400 Switch PLAYER@1FFF9164 from QUIET to NEXT on EVT_TIMER (5306 cycles in 50 ms)
4400 Switch PLAYER@1FFF9164 from NEXT to SOUND on ELSE (1 cycles in 0 ms)
`

Best

Jing

InfinityCode777, 2016-11-08:
Hi, @Mike58501, 

Thanks so much for the tips and I wish I could help but I am brand new to state machine. I am still learning it ... but I am following your post now and see if I could help some time later. 

Best

Jing

tinkerspy, 2016-11-25:
Hi Jing,

I finally got around to pulling my Teensy 3.1 out of 'that other project' and looking at this issue. Comparing the trace from the Teensy and an Uno made it clear that the Teensy was not firing the EOPAT event at the end of the pattern.

```
5200 Switch PLAYER@1FFF915C from SOUND to QUIET on EVT_TIMER (6278 cycles in 50 ms)
5250 Switch PLAYER@1FFF915C from QUIET to NEXT on EVT_TIMER (6321 cycles in 50 ms)
5250 Switch PLAYER@1FFF915C from NEXT to SOUND on ELSE (1 cycles in 0 ms)
```

Compared to this on the Uno:
```
5491 Switch PLAYER@265 from SOUND to QUIET on EVT_TIMER (756 cycles in 50 ms)
5573 Switch PLAYER@265 from QUIET to NEXT on EVT_TIMER (751 cycles in 50 ms)
5605 Switch PLAYER@265 from NEXT to REPEAT on EVT_EOPAT (1 cycles in 0 ms)
5684 Switch PLAYER@265 from REPEAT to START on ELSE (1 cycles in 0 ms)
```
The reason was simple. The Teensy uses 4 bytes ints while the Uno uses 2 byte ints. I changed the hard-coded int size into sizeof( int ) and now everything runs fine on the Teensy as well.

Thanks for reporting this!

Rgdz,
Tinkerspy

closed 2017-01-17 by tinkerspy