tinkerspy/Automaton #30: Using the "custom TrafficLight" stopping Serial Buffer in a timely way issue

Mike58501, 2016-10-22:
Without doing any surgery on the trafficlight .cpp & .h file I did some improvising on the sketch variations in the Automaton Documentation.

Below is INO code where my fertile imagination has gone to seeing if added Time Countdown Code can be also be run in a reactive event driven way.  Note: the Serial.flush () comment.  I tried to do due diligence research looking for a method or technique that I hoped would clear the buffer & found out Arduino Ide took that ability out after 1.01 version or so.  The only potential workaround (with my limited experience) that I have not tried yet is a crafted "for loop" within reassigned cursor positions on the lcd that keeps repeating lcd.clear () for that buffer run's position until that previous buffer runs out.  

In my defense,  I am thinking that it is plausible ... based on reliable sensors there may be a need to reset a time countdown on a traffic light possibly once & even twice.  What is a better fit to use than an event driven framework such as Automaton!? 
``` 
#include <Automaton.h>
#include "Atm_trafficlight.h"

Atm_trafficlight trafficlight;
Atm_button button_g, button_r;
Atm_timer firstTimeDown, secondTimeDown;
int test = 40;

void setup() {

  Serial.begin( 9600 );
  trafficlight.trace( Serial );

  trafficlight.begin( 7, 8, 9 ) // Pins 7, 8 & 9
  .automatic( -1, 8000, -1 ); // when
  button_g.begin( 2 )
  .onPress( trafficlight, trafficlight.EVT_GREEN ); //pin7 led1 on "kit on a shield"
  firstTimeDown.begin( 1000, test ) // Each step takes 1 second
  .onTimer( [] ( int idx, int v, int up ) {
    char buffer[50];
    sprintf( buffer, "Time left: %02d:%02d", v / 60, v % 60 );
    Serial.println( buffer );
  })
  .start();
  button_r.begin( 5 )
  .onPress( trafficlight, trafficlight.EVT_YELLOW ); //pin8 led2 on "kit on a shield"
  secondTimeDown.begin( 1000, 30 ) // Each step takes 1 second
  .onTimer( [] ( int idx, int v, int up ) {
    Serial.flush (); //this does not stop the first buffer from completing which is needed
    char buffer[50];
    sprintf( buffer, "Time left: %02d:%02d", v / 60, v % 60 );
    Serial.println( buffer );
  })
  .start();
}
void loop() {
  automaton.run();
}
    ```

Mike58501, 2016-10-26:
The above post said I could find only one potential solution to being able to stop a serial buffer from showing on an lcd screen after the code block has started to execute.  To hopefully not be guilty of a blind too much leading the blind ... I am now posting what appears to be another potential solution.  I modified to see what happens on the below "hands on"  example that came with the paid for "Kit on the Shield".  By usiing the ' Serial.end() ' function within the below non Automaton applicable Arduino code blocks; I am finding possibly another solution to being able to stop the Serial Buffer from continuing decrement ... so as to allow a needed reset with an entirely new ".repeat arg variable" of the TimeCountdown based on digital input.  Below is my first blush attempt ... but not certainly not conclusive especially in terms of making it do something similar enough specifically with the Automaton Framework ... which may require some kind of override???  On the upper instance of the  ' Serial.println ' execution ,  it allows one instance of the ' Serial.println ' to show on the Serial Monitor.  On the lower ' Serial.println '  the ' Serial.end() 'stops the ' Serial.println ' in the Serial Monitor completely

See below Serial.end() usage that I inserted ... I am not seeing this function in any of the Automaton Examples at this point in time.

~~~
//Push your buttons and see what happens...

const int  button1 =  2;
const int  button2 =  5;
const int  ledPin = 6;
int t = 10;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  pinMode (button1, INPUT);
  pinMode (button2, INPUT);
  pinMode (ledPin, OUTPUT);

}

void loop() {

  int  button1Press = 1;
  int  button2Press = 1;

  while (digitalRead(button1) == LOW) {
    if (button1Press  == 1) {
      Serial.println ("button1 Press");
      button1Press     = 0;
      playTones ();
      if (t == 10)
        Serial.end(); // stops 1 or both serial outputs
    }

  }

  while (digitalRead(button2) == HIGH) {
    if (button2Press  == 1) {
      Serial.println ("button2 Press");
      button2Press     = 0;
      playTones ();
      digitalWrite(ledPin, HIGH);
      delay(100);
      digitalWrite(ledPin, LOW);
      if (t == 10)
        Serial.end(); // stops 1 or both serial outputs
    }
  }
}

void playTones() {

  Serial.println  ("Playing 250hz Tone...");
  tone (A3, 250);
  delay(100);

  Serial.println  ("Playing 440hz Tone...");
  tone (A3, 440);
  delay(100);

  Serial.println  ("Playing 660hz Tone...");
  tone (A3, 660);
  delay(100);

  Serial.println  ("Playing 880hz Tone...");
  tone (A3, 880);
  delay(100);

  noTone (A3);

}

~~~

Mike58501, 2016-11-04:
TinkerSpy &/or anyone with enough experience with Automaton,

Since I have exhausted IMHO any possibility that "Serial.end ()" would work to clear the serial buffer to allow for a reset with the serial buffer actually cleared, I found this code at the following Arduino thread by Robin2: http://forum.arduino.cc/index.php?topic=288234.0  

Whether this can be used to clear the buffer & then restart new time countdowns with new updated .repeats in the ACTIONs code within an  Automaton State Machine is something that I & possibly others would really appreciate a bit of assistance on.  That is to hopefully  prevent too many people IMHO from trying to re-invent the wheel.  Possibly for an even more important project than the possible more robust TrafficLight Roundabout.  TrafficLight would IMHO would need more states than just Green, Yellow, & Red ... that's for sure ... IMHO.
~~~
void clearInputBuffer() {
   while (Serial.available() > 0) {
        Serial.read();
   }
}
~~~

tinkerspy, 2016-11-07:
Hi Mike,

I've run the code from your first post and it seems to run well on my Arduino Uno. It runs two independent timers, one counts down from 40 seconds, the other from 30 seconds and it simultaneously drives a traffic light with two button triggers. 

I see no problems with the serial output at all, it displays the counting timers and the traffic light trace() output on the Arduino console.

Using the single traffic light in a setup with multiple coordinated traffic lights would require another state machine as a controller that drives the lights by triggering their public events.

Output sample (I marked the different timers with Time1 & Time2 for clarity):
```
0 Switch TRAFFICLIGHT@2FC from *NONE* to GREEN on *NONE* (1 cycles in 0 ms)
Time1 left: 00:39
Time2 left: 00:29
Time1 left: 00:38
Time2 left: 00:28
Time1 left: 00:37
Time2 left: 00:27
Time1 left: 00:36
Time2 left: 00:26
Time1 left: 00:35
Time2 left: 00:25
Time1 left: 00:34
Time2 left: 00:24
Time1 left: 00:33
Time2 left: 00:23
Time1 left: 00:32
Time2 left: 00:22
```

Serial output buffers are beyond the control of Automaton. It just sends data to a Stream object when requested to by the trace() call.

Rgdz, Tinkerspy

Mike58501, 2016-11-08:
Hi TinkerSpy,

I do appreciate the below "heads up" statement which I am interpreting as needing to use Automaton's millis (interval variable) & counter (decrement repeat variable) directly within a custom machine very similar to your custom TrafficLight Machine with some tricky Giant Sized LCD time countdowns.  The triggers from the Timer Machine seem to fall short when it comes to the when of the "stop before finish" of the 1st countdown & the when of the new 2nd "not continued but revised" 2nd countdown & possibly even a 3rd revised countdown.  The revisions are based on digital Read statuses.

**_TinkerSpy quote:

"Using the single traffic light in a setup with multiple coordinated traffic lights would require another state machine as a controller that drives the lights by triggering their public events."_**

As I am getting farther along with Automaton ... I notice that I can control the "when" of a single Serial.println message happens by actually including it in the Enter Column or the Exit Column of the Custom State Machine's pertinent State or Row ('case' action within it.)  Of course a specific EVT based most likely on a "when" controlled EVT Trigger is also involved.  Bypassing the sketch portion almost completely.  Since I am writing to a 20,4 LCD ... this is nice because I can make the Cursor Positions overwrite what the previous "when" or State Actions Printed Out.

The time countdown code seems a bit trickier.  It seems by adding additional states IDLE, START, WAITMS, & TRIGGER to the existing custom TrafficLight .cpp & .h ... it would open the door to controlling the "when" of the first Time Countdown & even have an the ability somehow to use a terminator character with the STREAM in a state just before a 2nd Time Countdown starts its "when" process.  Contingent on DigitalRead (triggers) statuses the Time Countdown # of repeats may need to change which best is done by starting a whole new predefined (repeats) Time Countdown.  A specific analysed "when" state within the table causing it to happen.

You experienced the same problem as I did with the two processes simultaneously printing to the serial monitor when the goal is to have control over the "when" of the  terminating of the 1st Time Countdown process & control over the "when" of the start of a whole new 2nd Time Countdown.  The new updated 2nd Time Countdown process needs to replace not continue the 1st before the 1st is completely done.  Correct me if I am wrong but it seems that I was confusing the "buffer" with the whole countdown process when I not seem to think that the "buffer is more a per line proposition"???

Below mentions the above referenced "terminator character" that hopefully can be used within a custom state machine when the need arises:  [https://www.arduino.cc/en/Reference/StreamReadBytesUntil]

growing in my appreciation for the Automaton approach,

Mike

tinkerspy, 2016-11-08:
> The triggers from the Timer Machine seem to fall short when it comes to the when of the "stop before finish" of the 1st countdown & the when of the new 2nd "not continued but revised" 2nd countdown & possibly even a 3rd revised countdown.

I don't get what you're talking about. What do you want the timers to do? Do you want to run the timers in sequence? Nothing is easier than that.

```c++
#include <Automaton.h>

Atm_timer firstTimeDown, secondTimeDown;
int test = 40;

void setup() {

  Serial.begin( 9600 );

  firstTimeDown.begin( 1000, test ) // Each step takes 1 second
    .onTimer( [] ( int idx, int v, int up ) {
      char buffer[50];
      sprintf( buffer, "Time1 left: %02d:%02d", v / 60, v % 60 );
      Serial.println( buffer );
    })
    .onFinish( secondTimeDown, secondTimeDown.EVT_START ) // Start timer2 when timer1 finishes!!!
    .start();
  secondTimeDown.begin( 1000, 30 ) // Each step takes 1 second
    .onTimer( [] ( int idx, int v, int up ) {
      Serial.flush (); //this does not stop the first buffer from completing which is needed
      char buffer[50];
      sprintf( buffer, "Time2 left: %02d:%02d", v / 60, v % 60 );
      Serial.println( buffer );
    });
//  .start(); Don't start the secondTimeDown timer immediately!!! 
}
void loop() {
  automaton.run();
}
```

The way you configured it both timers are started at the same time. They will then continue running simultaneously. I changed it so that secondTimeDown is not started immediateliy (removed the start) but it's triggered by the firstTimeDown.onFinish() handler which is only executed when the first timer finishes its last repeat.

Buffer is just a temporary space to store characters in that are later printed by println(), it has nothing to do with the workings of the timers.

Mike58501, 2016-11-08:
TinkerSpy quote:

"I don't get what you're talking about. What do you want the timers to do? Do you want to run the timers in sequence? Nothing is easier than that."

Here is Mike's reply:

The need is not just the secondTimeDown to hold off starting until the firstTimeDown is finished.  That would be like you say a simple sequence.

The firstTimeDown needs to have somthing actually stop it from displaying BEFORE it is finished.   In my post I was hoping to be able to have that increased level of control from some added extra states & events in the trafficlight .cpp & .h file.   It seems like in my scenario the time countdown increment need would have to come from within the further customized trafficlight machine.

 The EVT trigger to the before actual time finished STOP of the firstTimeDown could be based on a digital machine input or an onPress from a button machine.  Because there will be slight variations of the "when" of the countdown stop event ... it is helpful for the secondTimeDown to function as a DISPLAY CORRECTION to the first.  I believe your  onFinish responds with the stopping of the display only when the firstTimeDown time is up.

So to sum up ... the firstTimeDown could get not as accurate if left just to onFinish without an "intervention" of stopping & restarting with an adjusted REPEAT variable.  That is because an Arduino pin status change  has a bearing on the "towards the end" actual time remaining.

Mike

tinkerspy, 2016-11-08:
> The firstTimeDown needs to have somthing actually stop it from displaying BEFORE it is finished. In my post I was hoping to be able to have that increased level of control from some added extra [...]

If you want to stop a Timer just send it an EVT_STOP event like this:

```c++
void setup() {

  timer.begin( 100 )
    .repeat( -1 );

  button.begin( 2 )
    .onPress( timer, timer.EVT_STOP );
}
```
> So to sum up ... the firstTimeDown could get not as accurate if left just to onFinish without an "intervention" of stopping & restarting with an adjusted REPEAT variable. 

Adjust the repeat by calling timer.repeat() like this:

```c++
void setup() {
  ...
  timer.begin( 100 )
    .repeat( -1 );

  button.begin( 2 )
    .onPress( []( int idx, int v, int up ) {
      timer.repeat( 3 );
   });
  ...
}
```

closed 2016-11-18 by Mike58501