tinkerspy/Automaton #38: Atm_digital and atm_button should use interrupts insted of digitalRead()

PeronistaUniversal, 2017-03-03:
Just checked the code and I think atm_digital & atm_button should use interrupts instead of polling the pin with digitalread() on the loop. I'd do it but I'm too a noob for toying with the code.

Thanks by the library btw, is great

tinkerspy, 2017-03-04:
Hi Peronista,

Actually the Automaton framework was designed so that you wouldn't have to use interrupts. Interrupts are one way to do multitasking, state machines are another. If Automaton would use interrupts it would be less portable and it would conflict with other libraries that use the same interrupts. 

That said, there's nothing to stop you from using interrupts in your own code, they'll work fine with Automaton.

Regards,
Tinterspy

Mike58501, 2017-03-04:
Just to share from an amateur's perspective ... I can see where Peronista is coming from.  Looking at some of the dialog on Arduino forums there are instances where polling of a bunch of pin states is not as good as using the pins as interrupts.  We have an Automaton program where we are making sure that the number of pin pulses from a Sparkfun coin mech's pulse output is put into the program's logic.  The chosen way was to use the serial option over the interrupt option for the coin mech.  It appears that serial communication uses interrupt ... but in a sllghtly different way.

The Automaton state machine approach seems to be able to make it so the polling only occurs when the logic is asking for seeing that event.  This may or may not explain why interrupts are less necessary with whatever kind of a state machine is used.

I too think that the Automaton Reactive State Machine approach has potential ... despite some naysayers.

closed 2017-03-26 by tinkerspy

matthijskooijman, 2019-01-10:
I was just about to report this request, when I found this issue. I would think interrupts would be interesting as an optional feature. AFAICS button & digital could probably be modified to handle changes in an ISR rather than by polling (might need some additional states to prevent too much handling inside the ISR).

The usecase I have is sleeping: when the only active state machine is a button state machine, and it is activated using an ISR, you could put the MCU to sleep and have it woken up by the pin interrupt. When the state machine is waiting for a pin change, it could be in a sleeping state. Then, you can put the MCU to sleep when all state machines are sleeping.

I think a similar argument could be made for timers, which could be handled using interrupts rather than (or in addition to) polling to make sure the MCU can sleep while waiting for a timer.

An alternative take on this would be to keep everything polling, but somehow let automaton expose a list of events it waits for. If this list only includes pin changes and there are interrupts enabled for these pins (even with empty ISRs), the MCU could sleep, and polling would resume after wakeup. Similarly, if timers would be somhow centrally recorded, a sketch could ask Automaton for the nearest timer and sleep that long.

I guess this needs some more thought, and maybe an issue of its own...

tinkerspy, 2019-01-11:
That's an interesting thought, but I still don't think Automaton should incorprate interrupts. I've given some arguments above. There's nothing to stop you from doing what you want outside of the library configure an interrupt handler and put the mcu to sleep. When the mcu wakes up so will Automaton. A button press will then be picked up immediately.

I've used a pin change interrupt inside an Automaton state machine in the past, but the use case then was that I needed to respond to pulse width changes on that pin very quickly.

matthijskooijman, 2019-01-11:
Thanks for your thoughts. I'll indeed see if I can make this work from outside of Automaton, or perhaps with a customized version of button. Might be a while, though.

tinkerspy, 2019-01-11:
I'd like to see the result if you have something. I've never done anything with mcu sleeping but it always sounds like a cool way to make battery powered devices.

dvv, 2020-04-16:
Hello, @tinkerspy!

I would like to make `Atm_button` virtual in sense that it could be able to read "pin" state from a callback.

Then that callback would reduce to `digitalRead(pin)` for physical buttons. Those who want interrupts might set that callback to read a volatile flag set inside their interrupt handler.
I personally would convert a resistive keyboard (a car steer buttons) voltage to multiple pseudo-binary buttons.

Do you think it deserves to be in master or shall I go clone `Atm_button.?pp`?
If the latter, how could I push a "pin" state from outside the `Atm_button`?

TIA

tinkerspy, 2020-04-21:
Best make a separate library. I'll link to it when you're done. I fear interrupts will make for less portable code.

You could use atm_comparator for the resistive steering wheel buttons I think.

dvv, 2020-04-22:
What I mean is just to not tie to digital pin state as button state and to allow kinda the following:
```
int onReadButtonState() {
  return digitalRead(0);
}
b0.begin(onReadButtonState).debounce(...).onPress(...).onRelease(...).longPress(...);
```

How they will be managing the return of `onReadButtonState` is then completely out of the library scope.
There's also the need just not to setup the physical pin in the case of read helper.

Also, seems you yourself were thinking of something like that -- https://github.com/tinkerspy/Automaton/blob/master/src/Atm_button.cpp#L3

On atm_comparator: then I lose the click/longclick events and have to reimplement them by myself as I can not _drive_ atm_button state machine with atm_comparator changes, right?

tinkerspy, 2020-04-22:
Yeah, that's probably a good idea.
I'd make it a separate method, not put it in begin().

dvv, 2020-04-22:
It's completely up to you. TIA