tinkerspy/Automaton #39: D13 on AT328P becomes unusable when automation is included

jamesarm97, 2017-03-09:
I am using the D13 pin on the 328p to drive a FET that drives a speaker. This is shared with the onboard LED and has been working fine. As soon as I include this library (without even defining or using any part of it, the D13 pin becomes useless. The onboard led barely blinks when I am toggling it to control the buzzer. I thought the pin was dead but if I just comment out the Automation.h file in the main INO and recompile the pin starts working. Is some portion of the code initializing the D13 pin as an input instead of output or enabling SPI or a shared function that would conflict even when I have not defined or called any portion of the library?

tinkerspy, 2017-03-10:
What version of Automaton are you using? I believe there was a conflict with constant naming in earlier versions that should be solved if you use the latest build from Github.

jamesarm97, 2017-03-10:
I will check tomorrow. It was installed via library manager and not downloaded directly from github.

tinkerspy, 2017-03-10:
Strange, the library manager version should be up to date enough.

Could you try this with 'automaton.h' included and excluded?

```c++
Serial.println( D13 );
```

jamesarm97, 2017-03-10:
D13 is not defined. I have just been using pinMode(13, OUTPUT) to set it as an output then digitalWrite(13, HIGH) or LOW. with Automation.h commented out the onboard D13 LED is bright and blinks normally and the FET on the same pin driving a buzzer works. With it enabled the onboard D13 led shows dim and buzzer not working because it isn't driving hard on or off. I looked at it with a scope and can see the difference.

jamesarm97, 2017-03-10:
Adding automation.run() in the loop does same thing, but if I add Atm_led led1 then led1.begin(13).blink(100,100) and trigger it then it works. So something in automation is changing the pin 13 from an output to something else when included even though I have pin 13 set to an output.

tinkerspy, 2017-03-10:
Could you try this:
- Include automaton.h so the problem occurs.
- Remove the includes in the lower section of automaton.h and see if the problem goes away.
- If it does then restore them one by one until it appears again.

Rgdz, Tinkerspy

jamesarm97, 2017-03-10:
It shows 1.01 in the arduino library manager but the dropdown select version to install only goes to v1.0. The library.json file shows 1.01

jamesarm97, 2017-03-10:
Ok, new problem or just don't understand why it wouldn't work this way. 

Atm_led buzzer;
buzzer.begin( 4 );
buzzer.on() or buzzer.trigger( buzzer.EVT_ON ) does not do anything, actually makes the led on the same pin really dim.
yet:
buzzer.begin(4).blink(100, 100);
buzzer.on() works? Why cant you define an led / digital output then turn it on without having to call blink (in this case I do not want it to blink, just turn on).

tinkerspy, 2017-03-11:
That's all very strange, you're saying that this minimal sketch (let's talk full sketches to avoid confusion) doesn't turn a led connected to pin 4 (with its anode) on?

```c++
#include <Automaton.h>

Atm_led led;

void setup() {
  led.begin( 4 );
  led.on();
}

void loop() {
  automaton.run();
}
```

That sketch should just do:

```c++
pinMode( pin, OUTPUT );
digitalWrite( pin, HIGH );
```

What development board are you using?

tinkerspy, 2017-03-15:
Any news, James?

Rgdz, Tinkerspy

closed 2017-03-26 by tinkerspy