tinkerspy/Automaton #41: tone/noTone is not available for Due

vermut, 2017-04-22:
https://travis-ci.org/MysteriaLV/boxes/builds/224787770

Maybe Atm_player.* shouldn't be included at all when `#ifdef _VARIANT_ARDUINO_DUE_X_` ?

P.S. Love your lib. Real life saver. Thanks.

Mike58501, 2017-04-23:
That is interesting that TinkerSpy's Automaton is being used some for Arduino Due.  Has anyone used & tested Automaton for the Arduino MEGA 1280 2560?  The Mega certainly could make lack of memory problems less likely ... if there are not other issues besides just paying attention to the extra IO.

rawtaz, 2017-04-23:
What's so interesting about it?I use Automaton on a Mega 2560 because it has more memory than the Leonardo. It works great.

tinkerspy, 2017-04-23:
It should work on all Arduino supported platforms, at least that's what I aim for. It works fine on the teensy range or on the esp8266 as well.

Vermut, is that the Duemilanove I see in the Arduino IDE? Because it seems to compile fine with that setting. Atm_player has more functionality than just controlling a speaker, so I would prefer to just conditionally disable the (no)tone command if that would be necessary.

vermut, 2017-04-23:
Duemilanove and Due are different things :) I also got confused first.

Here is a short disambiguation: http://platformio.org/boards?filter%5Bframeworks%5D=arduino&filter%5Bname%5D=due

conditionally disabling the (no)tone is fine. 
or even better -  emulating it: http://forum.arduino.cc/index.php/topic,136500.0.html

tinkerspy, 2017-05-18:
Hi Vermut,

I finally implemented the ifdefs in the master branch.
Could you let me know if it fixes your problems.

Sorry for the delay.

Regards,
Tinkerspy

vermut, 2017-05-29:
Still failing. You sure *ifdef* is correct pragma? Seems like you disabled _tone_ for all platforms except DUE.

tinkerspy, 2017-05-31:
Yeah, looks like I got the conditional the other way around. Now, that never happened to me before ;-)

Is it OK now?

vermut, 2017-05-31:
Yes, much better: https://travis-ci.org/MysteriaLV/boxes/builds/234074868

vermut, 2017-07-03:
Please bump library.json so Platformio will pick up new changes.

rfestag, 2017-11-20:
I'm running into the same issue with my Bluefruit feather (ARDUINO_FEATHER52).

tinkerspy, 2017-11-20:
I fixed it in the latest master, can you check if it works for you?

Rgdz, Tinkerspy

rfvermut, 2017-11-20:
Many library managers are not noticing those changes. Please bump the version - #46

closed 2017-12-06 by vermut