tinkerspy/Automaton #43: automaton.run() called from timer

jej, 2017-06-28:
Hi,
Is it possible to call automaton.run() in a function called by timer? (with Ticker library) I would like (for example) leds running when the loop() is not called.
I tried it with a timer called every 1/1000s. My sketch works mostly well for the led (even during wifi setup) but when a button is pressed (managed by automation), it stops...
Cheers.

tinkerspy, 2017-06-28:
Hi Jej,

Sure, whenever automaton.run() is not called by the loop() you can call it from other places. All it does is call the different state machine's cycle() method in turn. Are you using an esp8266 by any chance? There's an Automaton library for that. It's weird that it would stop when you press an (Atm_button?) button. That should never happen. Atm_button doesn't block.

Regards,
Tinkerspy

jej, 2017-06-28:
Yep I use esp8266, but I don't really need the http and wifi machines.
I change the parameters of my machines after events, and maybe the problem occurs if the interrupt launch the run() meanwhile... That's why I was doubting that's possible to run this function in parallel. When launched in the loop(), it's sequential with my code.
Thanks for you answer.

Mike58501, 2017-06-28:
Correct me if I am wrong ... but in general "running whatever in parallel ability" seems to be one of Automaton's weaknesses.   I found that sequential operations were much easier to come up with.

closed 2017-06-29 by jej