tinkerspy/Automaton #44: Led.begin() hangs on Leonardo

vermut, 2017-07-07:
Basically the simple sketch like this
```
Atm_led led1;
void setup() {
	led1.begin(2);
}

void loop() {
	delay(1000);
}
```

results in Leonardo hanging indefinitely. Worked fine in Due.

I found out commenting out this line makes it fall through: https://github.com/tinkerspy/Automaton/blob/master/src/Atm_led.cpp#L32

Is there any dark magic there? I cannot even find a `state()` function for Led.

tinkerspy, 2017-07-08:
Dark magic? Well, not entirely, but there's something happening
underneath the surface...

Atm_led is a class that is based on the Machine class. It's the Machine
class that has the state() method in it. Any class based on Machine
inherits that method, but it may override it.

Objects (like your led1) based on the Machine class are automatically
added to the 'automaton' scheduler class. The 'automaton' scheduler is
responsible for calling the object's cycle() method (also inherited from
Machine) that let's the object do its work. For the scheduler to work it
must be called from the loop function. Oh, and, no delays!

It seems unlikely that your sketch would run on a Due (that would be black magic indeed). To make it work I added ```automaton.run()``` to your loop() function so the scheduler actually gets called. Also, the ```led.begin(2)``` initializes the led object and attaches it to pin 2, but it doesn't do anything. To make the led do something you have to give it a command like ```led1.on()``` (to turn it on) or ```led1.start()``` (to make it blink). Until then it just sits there awaiting your commands. 

```c++
#include <Automaton.h>

Atm_led led1;
void setup() {
  led1.begin(2);
  led1.start();
}

void loop() {
  automaton.run();
}
```

Mike58501, 2017-07-08:
Is there a way to successfully add an if conditional statement along with the 
~~~
automaton.run();
~~~
within the loop() function?

It seems like that could be helpful to work around the somewhat rigid sequential nature of commands done in setup that are picked up by the 'automaton' scheduler.  That is possibly adding a unique looping if conditional statement that would possibly run concurrently or in parallel with the automaton sequential process???

tinkerspy, 2017-07-09:
There's no reason you couldn't add an if statement in the loop(), as long as you don't do anything blocking in it. It's not really automaton-like, but it would work fine. The run() method executes a slice of each registered state machine and then exits. So any extra code in the loop statement would execute with the same frequency as any state machine.

```c++
void loop() {
  automaton.run();
  if ( <some condition> ) led.start();
}

Mike58501, 2017-07-09:
We appreciate your answer.  Can you give us an example where it would it make sense to add a similar  if  statement in the LOOP COLUMN of your Machine Template Editor?

tinkerspy, 2017-07-09:
1. The loop column is not in the Machine editor but in the Machine templates it writes.
2. In my experience it is rarely necessary to use the loop column apart from placing ATM_SLEEP in it so that the state becomes a sleeping state.
3. The by placing an action ID in the loop column you can put code in the action handler that gets called on every cycle that the state machine is in that particular state.

vermut, 2017-07-09:
Wow wow. You hijacked my very important thread :)

Ok, my fault, I oversimplified the example. Problem is not that my led is not working. Problem is the whole board freezes waiting in forever `while` at https://github.com/tinkerspy/Automaton/blob/master/src/Atm_led.cpp#L32

Here is slightly extended example where I tried to make my point more clear:
```
Atm_led led1;

void loop() { 	delay(1000);   }

void setup() {
	pinMode(LED_BUILTIN, OUTPUT);
	digitalWrite(LED_BUILTIN, LOW);   // LED will turn off
	delay(100);

	led1.begin(2);
	digitalWrite(LED_BUILTIN, HIGH);    // ... and never turn on, never reaching this
}

```

tinkerspy, 2017-07-10:
Hi vermut,

It's always best to provide a minimal but complete example that illustrates the problem ;-) You're right, this shouldn't happen. I don't have a Leonardo, but I have an old teensy 2.0 which uses the same Atmel chip. I'll see if I can replicate this...

I'm kind of busy with my Automaton based event-driven quadcopter, though. The kids are getting impatient ;-)

Rgdz,
Tinkerspy

vermut, 2017-07-10:
Ok, this is low-medium priority for me. No Leonardo's in production yet. But would be nice if you take a look.

tinkerspy, 2017-07-13:
Hi vermut,

I just tried your sketch on my Teensy 2.0 (also with a ATmega32u4 microcontroller on board) and it ran fine. I don't have a Leonardo to test it with. I tried to select 'Leonardo' in the IDE and upload it that way, but apparently the boards aren't that compatible. Perhaps you can put some trace statements in the Atm_led::begin method and figure out where the problem occurs?

Rgdz,
Tinkerspy

vermut, 2017-07-13:
I told you, I found out commenting out this line makes it fall through: https://github.com/tinkerspy/Automaton/blob/master/src/Atm_led.cpp#L32

tinkerspy, 2017-07-13:
Oh yeah, you did, sorry. 
Is it the state() of the cycle() call that hangs? 
Or does state never return 0? 

Rgdz,
Tinkerspy

vermut, 2017-07-13:
`state()`, as I mentioned in first message. And asked "Is there any dark magic there?"

What is `state()` waiting for which is different in Leonardo? Some non-existent timer? Some non-orthodox interrupt?

tinkerspy, 2017-07-14:
No idea what's different. Every machine's begin calls the parent Machine class' begin method which sets the 'current' value (which holds the current state) to -1 and the next value to 0. 

Then the child machine cycles itself until the cycle method (noticing that next != -1) switches the current state from -1 to 0 and executes the 0 state initialization code if it has any. The real problem is probably in the cycle() method. 

The state tables are stored in the code segment with the PROGMEM directive (RAM is precious on AVR's). Cycle has to do some voodoo to read that, in the read_state macro (defined in Automaton.h). Could that perhaps be giving you problems?

I've taken special care to avoid interrupts and timers in the main Automaton library and stuck to the standard Arduino library exactly to avoid compatibility problems. It should run everywhere. But the memory juggling cannot be avoided. If the state tables were kept in RAM memory would run out very quickly on many Arduino targets.

vermut, 2017-09-05:
Well, either I was stupid or recent lib update fixed something.
Can't reproduce anymore. Blinks like a baby.

closed 2017-09-05 by vermut

tinkerspy, 2017-09-05:
Thanks for the heads up!