tinkerspy/Automaton #45: Best way to handle MCP23017-pins with LED-machine?

gtom, 2017-07-16:
I would like to use an led-machine (only EVT_ON and EVT_OFF needed) with 16 pins controlled through mcp23017. I'm using Adafruit-MCP23017 library:

```
#include "Adafruit_MCP23017.h"
Adafruit_MCP23017 mcp1;

// in setup:
mcp1.begin( 1 ); 
  for ( int i=0; i<=15;i++ ){
    mcp1.pinMode( i, OUTPUT );
}

// somewhere in loop:
mcp1.digitalWrite( 6, HIGH);

```
I'm thinking about building an own machine derived from led-machine to handle different way of initializing and writing to pins. But before I start: is there a better way to achieve this? 
gTom

tinkerspy, 2017-07-17:
First you need to be clear about what you want to achieve. If you just want to switch the individual leds on and off without a time element involved (like with blinking or fading) then it's questionable that you need a state machine at all. 

It might be nice to have an Automaton-like interface to the leds, but you could achieve that even with a dummy state table. The machine would basically be sleeping all the time.
 ```c++
const static state_t state_table[] PROGMEM = {
    /*               ON_ENTER      ON_LOOP  ON_EXIT  ELSE */
    /*    IDLE */          -1,   ATM_SLEEP,      -1, IDLE,
};
```
You could then create an interface like this:

```c++
mcp1.begin( 1 );
mcp1.trigger( EVT_LED1_ON );
mcp1.trigger( EVT_LED2_ON );
mcp1.trigger( EVT_LED2_ON );
mcp1.on( 1 );
mcp1.off( 1 );
```

By just implementing your own trigger(), on() and off() methods.

gtom, 2017-07-17:
Thanks, I will think about it. When I need some timing within this, I think I will try to setup a new machine.
gTom

closed 2017-07-17 by gtom