tinkerspy/Automaton #47: Accessing Non-Static Class methods

johnmastri, 2017-10-18:
Apologies if this is not the place to post this, but I've been caught up on this for quite some time:

https://stackoverflow.com/questions/46802364/invalid-use-of-non-static-member-function-arduino-automaton

rawtaz, 2017-10-18:
You are trying to call a function on something named `sensor` - where does that `sensor` come from? I don't see it anywhere in the code.

euphi, 2017-12-09:
What you need is a solution for #17.
After 1 1/2 years it seems that I now face a problem where there is no elegant solution without a lambda capable of using a captured this.

euphi, 2017-12-09:
@tinkerspy calling callbacks of an object (class instance) is an usecase that happens quite often. 

I currently have a situation where I want to do a http-request if a specific state is entered and then switch state when the result is received - with different events depending on the result. So the http-client (Atm_esp8266_httpc_simple) is member of the main state machine and I want to have a callback to one of the member functions that reads the result and then switch state.

A generic solution would be to allow real lambda function for connectors:
I finished to implement a proof of concept of #17 - but it does not work. My approach to store a pointer in atm_connector to atm_cb_lambda_t inside the union does not work because it points to a std:function object that is deleted (it is created in local scope, so it result in exceptions). It is also not possible to store the atm_cb_lambda_t inside the union, because it is of "non-trivial type".
Maybe this could be solved by storing the atm_cb_lambda_t outside the union (but still inside the atm_connector). However the necessary refactoring to atm_connector seems to be to complex for me for now. :-(

A less generic, but very nice solution for my specific problem is to introduce new callbacks for the Atm_esp8266_httpc_simple class depending on the return code.

euphi, 2017-12-09:
I pushed my try to github, see https://github.com/euphi/Automaton/tree/test_for_lambda .

As said, this does not compile. If you change it to store a pointer, it compiles, but crashes.