tinkerspy/Automaton #51: Encoder switch debounce

jfitter, 2017-11-28:
The encoder machine needs to implement a debounce. When used with mechanical rotary encoders an arbitrary number of events are fired when the encoder is rotated. In my hardware implementation I have a high quality Bourns encoder and capacitive filters on the switches, but still receive 2 to 6 events for each incremental encoder position.

DrDiodac, 2018-07-28:
I have similar problems like jfitter. But when I run sketch just with state encoder machine and add some of leds all work very well. But when I try any implementation to other things coming problem, machine working unstable, reading few positions at once. I have quality encoder, hardware debounce and keep attention to detail  when I write code. I working on Teensy 3.2.

knifesk, 2022-09-17:
+1 here. it fires once up and down at least no matter the direction