tinkerspy/Automaton #53: Atm_opendrain welcome?

dvv, 2018-01-21:
I have a (set of) active-low relay which I control both via a controller pin and an external switch.

That controller pin carries two tasks:

- I can get the relay state via `((DDRX & _BV(Y)) || !(PINX & _BV(Y)))`
- I can set the relay state via `if (on) DDRX |= _BV(Y); else DDRX &= ~_BV(Y)`

So I'd like to have let's-name-it `Atm_opendrain` which is `Atm_led` without PWM parts and overridden `on/off` and `Atm_digital/Atm_button` with custom `state` reader.

How would you suggest to realise this?

TIA

tinkerspy, 2018-01-22:
> thread[2].>

Perhaps you could use a combination of an Atm_bit object (to control the
relay) with an Atm_digital object to read the state. From the
Atm_digital's onChange() handler you can set the Atm_bit object's state.
The Atm_bit's state method then should both reflect and control the
relay's state.
You can of course write a custom class to control and read the relay.
I've used a latching relay in the past which I wrote a custom state
machine for (pre-Automaton).
Rgdz,
Tinkerspy

Links:

  1. https://github.com/tinkerspy/Automaton/issues/53
  2. https://github.com/

dvv, 2018-01-22:
Thank you. I would like to merge `Atm_button`'s debouncing and `Atm_led` blinking (in sense of limited on state duration mostly). Am not afraid to write custom class, but am way far from making own state machine.

euphi, 2019-01-10:
It's almost one year, but if you are still intested have a look at my PR https://github.com/tinkerspy/Automaton/pull/50/ . It allows to use C++ inheritance to have on functions for setting the output. So you can just derive Atm_opendrain and overwrite the `switchOn()` and `switchOff` functions.