tinkerspy/Automaton #57: LED machine for addressable leds (neopixel)

sliptonic, 2018-05-19:
Since these things are so widely used, I thought somebody must have already written some code using them with automaton.  I haven't found anything yet.  Can you point me to any example using either the adafruit neopixel library or FastLED?

tinkerspy, 2018-05-22:
I haven't heard of one. I ordered an APA102 strip recently so I'll probably cook something up for that. I'm thinking I'll build a low-level machine that updates the strip whenever anything has changed and higher level 'Atm_led' type machines that turn individual leds on and off.

antonine, 2018-11-26:
Any chance for a follow up on this topic. Can a newby be of any help here??

tinkerspy, 2018-11-26:
I'm working on digitizing a seventies electromechanical pinball machine and a APA102 LED strip plays a big part in that project. The code contains a rough apa102 state machine. You can find it here if you want to play with it:

https://github.com/tinkerspy/pinball

It doesn't really fit the standard led animation model, I think. If you can write something better suited to that (perhaps by using the fastled library) that would be great.

hjmhardsoft, 2023-02-22:
hi @tinkerspy congratulations by your Automaton Library, is fabulous!!!

How are you? all fine?

tinkerspy, 2023-02-22:
Thanks, I'm fine thank you. Though I left most of my Arduino hardware behind and moved to a Caribbean island...

hjmhardsoft, 2023-02-22:
Wow.  Very surprised by the radical change in your life.  Without knowing you, it struck me that after writing such good c++ code in these state machine libraries, after the pandemic you didn't make more commits.  But if you are well, welcome.  the improvements in this bookstore will be missed 😃

antonine, 2023-02-22:
Fully agree - excellent programming quality, good stuff

On holiday, from the largest Carribean island, best regards

Jan Wilms

tinkerspy, 2023-02-23:
I did build a pinball control system (check my other repo's) which used a LED matrix made of several addressable LED strips to control everything. Automaton proved very useful there as well. The whole pinball game logic is programmable in a byte code interpreter that runs inside Automaton. Lots of hardware in that project too.

[image]