tinkerspy/Automaton #58: Inverted Buttons

ddownn, 2018-07-12:
I am using a hardware debouncing circuit like this using a Hex Schmitt-Trigger Inverter:

[image]

As a result, the buttons are effectively inversed with the "Automaton button machine." 

What would be the best way to account for the inversion caused by the SN74HC14 in Automaton code so that button presses and releases can be recognized correctly? 

Is there some way to just flip what defines a button as "Pressed"

tinkerspy, 2018-07-12:
No, not in the button machine. But the Atm_digital machine provides the same basic functionality and also has an activeLow setting that should provide what you want. It has more flexibilty in the pin handling than Atm_button, you can also enable and disable pullups and software debouncing.

ddownn, 2018-07-12:
Ah, yes. Thank you

closed 2018-07-12 by ddownn