tinkerspy/Automaton #60: Posibility to specify pinmode on Atm_button

mbariola, 2019-01-02:
Hi,

I'm currently evaluating your framework to try it out on a Cortex M0 and an ESP32, both of which have INPUT_PULLDOWN capabilities (which is more suitable to my project).

I'll quickly hack it locally as my time is limited, but maybe you would consider extending Atm_button to provide an extra parameter for the PINMODE, for a minor update?

Best

tinkerspy, 2019-01-02:
Hi Best,

I could fairly easily add an input_mode argument to the begin() method, if that works for you. But Atm_button would still be 'active LOW'. Is that useful?

You could also use Atm_digital instead, that class can handle active high inputs. Just call pinMode after initialization:

~~~
Atm_digital button;

button.begin( 3, 20, true, true );
    .onChange( led, led.EVT_TOGGLE ); 

pinMode( 3, INPUT_PULLDOWN );
~~~

BTW: You just saved me 5 pulldown resistors on my current Teensy 3.5 project. :-)

mbariola, 2019-01-02:
Hi,
Actually I'd need the button to be active HIGH... Thanks for pointing that
out. Yes I much prefer working on pull-down and high active to save
resistors.
Maybe an option to configure the active state would be also needed.

I'll have a look at atm_digital, to see if it has the features I need..
Such as debounce.
Thanks for the quick answer and good luck with your project!

tinkerspy, 2019-01-02:
It's got debounce, just the fancier stuff like longpress, repeat & autopress are missing.

mbariola, 2019-01-02:
Seems good then, I'll try that way.
Just curious... Saved 5 resistors, anything to do with a wearable? 😉

tinkerspy, 2019-01-02:
No, a pinball machine. Not very wearable  ;-)

mbariola, 2019-01-02:
Nice! I'd love to see its tutorial for the automaton part 😉

tinkerspy, 2019-01-13:
OK, that sounds like a plan. I made a branch with the new pinMode() & active() methods. Note that they must be called *after* begin(). Would you mind checking it out and testing it?

~~~cpp
  button.begin( 13 )
    .pinMode( INPUT_PULLDOWN )
    .active( LOW )
    .onPress( led, led.EVT_ON );
~~~

mbariola, 2019-01-25:
Hey @tinkerspy !
Sorry if it took me a lot to answer, I will try the new branch with a brand new prototype starting this monday, and will let you know. Thanks for taking the time :-)

sjernigan, 2024-02-06:
I added this in fork https://github.com/sjernigan/Automaton and put in a PR