tinkerspy/Automaton #67: Graphwiz

mnolletti69, 2019-04-18:
Hi, the code and the web Machine Editor is really cool.
Please consider to add to the editor an automatic generated state machine picture.
It should be very easy with [https://www.graphviz.org/](url).
Many thanks.

tinkerspy, 2019-04-18:
Sounds cool, but I doubt I'll have the time.

The [source](https://github.com/tinkerspy/Automaton-Machine-Editor) is available on Github so you could give it a go yourself...

Mike58501, 2019-04-18:
mnoletti69,

No offense ... but since you suggested how easy it would be ... is it possible you would be ambitious enough to volunteer.  Look at it this way ... you would gain even more proficiency at the "easy" presentation tool.  I can't speak for tinkerspy but he might help out with your contribution on small portions of it ... when defining & using abbreviations etc.

mnolletti69, 2019-04-18:
Thank you, for your reply.

I'm the CEO of a small company and at the moment I spend all the free 
time for this project.

I appretiate the quality of your Automaton and give you an idea on how 
to improve it.

Try to past the following code (atm_button states) on the following link 
http://www.webgraphviz.com/ and look at the result:

digraph G {
  IDLE -> LIDLE
  IDLE -> LWAIT
  IDLE -> AUTO_ST
  WAIT -> PRESSED
  WAIT -> IDLE
  PRESSED -> REPEAT
  PRESSED -> RELEASE
  REPEAT -> REPEAT
  REPEAT -> RELEASE
  RELEASE -> IDLE
  LIDLE -> LWAIT
  LWAIT -> LPRESSED
  LWAIT -> LIDLE
  LPRESSED -> LRELEASE
  LPRESSED -> WRELEASE
  LRELEASE -> LIDLE
  WRELEASE -> LIDLE
  AUTO_ST -> IDLE
}

All you have to do is to generate the DOT text file (likewise your xml 
inline code), pass it to an executable (or an PHP function) on the web 
server (server side) and link the generated image.

If you search on the web you will find a lot code using Graphwiz (note 
that I have nothing to do with the developers).

For the moment I cannot help you anymore.

Kind regards

     Marco

PS I'm aware that depending on your webserver facilities, running an 
executable could be a pain.

tinkerspy, 2019-04-18:
"I'm the CEO of a small company and at the moment I spend all the free time for this project."

Hmm, know what you mean. I'm the CTO for two small companies, have an almost full time job on the side, a hobby project crying for completion and on top of that I'm packing for a family holiday.

Thatnks for your suggestion, it looks promising! I hope to get around to it.

klaudiusz223, 2019-10-13:
Hi,
Here is simple solution using XSLT (PoC). Tested only with few machines.
https://gist.github.com/klaudiusz223/006b1e61aefa9ad913bc4a3346985f05
How to  test this:
Use this website https://www.freeformatter.com/xsl-transformer.html#ad-output
put ATML definition into XML Input
put XSLT definition (given above) into XSL Input
push Transform XSL in new window

take result and put here
https://dreampuf.github.io/GraphvizOnline/

Atm_player example https://tinyurl.com/yyfvtl3y