tinkerspy/Automaton #68: Triggering event from inside Machine::action causes silent crash on Arduino Mega

00benallen, 2019-11-26:
Here's the offending code:
```
void Atm_machine::action( int id ) {
  switch ( id ) {
    case LP_STATE:
      otherMachine.event_function();
      return;
    case OTHER_STATE:
      return;
  }
}
```

My Arduino just silently restarts when I do this.

tinkerspy, 2019-11-26:
In general, you shouldn't be calling routines from one Machine object inside another machine object. I also wonder what event_function's relation to event() is.

If you do things you shouldn't do, it's not that surprising that things start crashing.

klaudiusz223, 2019-11-26:
I do almost the same, but i use pointer to another machine.
Event method in another machine is declared as public.

```
void Atm_machine::action( int id ) {
  switch ( id ) {
    case LP_STATE:
      otherMachine->event_function();
      return;
    case OTHER_STATE:
      return;
  }
}
```

00benallen, 2019-11-26:
@tinkerspy I did come to the same conclusion, however, I wonder if it would be possible to have a warning/error message.

Part of the problem was I couldn't find very good documentation on inter-machine communication in the first place and had to read a lot of source code to figure out a better approach. 

With neither an error message or good docs, my developer experience was "why does my arduino keep restarting?"

tinkerspy, 2019-11-26:
On microcontrollers warnings and messages are not really possible. You can't really start outputting strings on some usb/serial port in the hope that somebody's watching. An mcu doesn't really have a user interface until you write one.

Machine communication is documented in the Wiki:
https://github.com/tinkerspy/Automaton/wiki/Machine-communication

Even more documentation can be found in the second Machine building tutorial, when you can read how to create a connector (the outgoing part of event communication) that can be used to trigger other machines:

https://github.com/tinkerspy/Automaton/wiki/Machine-building-tutorial-2

You see, the idea is that machines call each other's triggers to communicate.

00benallen, 2019-11-27:
The machine communication part of the wiki doesn’t explain anything about
connectors, and the machine building example 2 doesn’t explain everything
about using push, since it’s just 1 example and it’s a simple one.

I get what you’re saying about warnings, although I do use some sensor
libraries which do output warnings in certain situations.

I wonder if the fact that I’m having this problem is an indication that the
docs need some additions? I get it now but it wasn't obvious until I spend
a while reading your source code.

tinkerspy, 2019-11-27:
It's not a trivial concept. If you've just freshly figured it out, perhaps you can explain it better than me. If you write a text (in markdown preferable, perhaps as a github gist) I'll add it to the wiki.

00benallen, 2019-11-27:
Cool okay

00benallen, 2019-11-28:
@tinkerspy Here's my first draft, feel free to edit!

https://gist.github.com/00benallen/a94b0919bcda986d90041c287bd9539b

00benallen, 2019-11-30:
@tinkerspy bump

00benallen, 2019-11-30:
@rawtaz why the thumbs down?

rawtaz, 2019-11-30:
@00benallen Nothing serious, but the author of this framework is a busy man and he's developing this library free of charge and in his free time. Bumping a thread just two days after your last message is not constructive - he will look at your contribution when he can, there's no need to remind him about it. It also pollutes the thread with that type of messages. In other words, give him some slack. That's why I thumbed down :)

tinkerspy, 2019-11-30:
Thanks, 00benallen, it's a lot better than my text. I think I'll just replace my text with it if that's alright with you. (I'll credit you, of course)

00benallen, 2019-12-01:
Yeah sure, I'm happy to have helped. Thanks for being open-minded! :)

tinkerspy, 2019-12-02:
It's on line. Thanks again!

00benallen, 2019-12-12:
@tinkerspy well I guess this is solved, I'll close this now

closed 2019-12-12 by 00benallen

Tsjompie, 2019-12-20:
Gents,

What would be my best alternative passing a textstring using this inter-machine communication?
The current option seems to use int's  only `void Machine::push( atm_connector connectors[], int id, int sub, int v, int up )`
Or is my  best alternative to use a global variable for that?
Thanks :)

tinkerspy, 2019-12-20:
Call a method from the callback that returns the string. When the callback is triggered it requests the string from the machine that called it.

Rgrdz,
Tinkerspy

Tsjompie, 2019-12-24:
OOps, still haven't been able to build as suggested above and kindly request for a little more directions on that (sorry).

From the ino, in which I have this: 

```
nexSerialStream.begin( nexSerial, nexSerialStream_buffer, sizeof( nexSerialStream_buffer ) )
    .list( nexSerialStreamlist )
    .trace( Serial )
    .onRxcommand( nexSerialStream_callback );
}
```

I'm able to execute a function `onrxcommand.push( lookup( 0, commands ) );` and can run code based on what comes in via the Rx. However I still fail in injecting a string like `n0.val=88` to use the `ENT_SEND_SET` or `get n0.val` using `ENT_SEND_GET`. 

The first function below works, do I need to 

> add the method from the callback that returns the string

 in the second `ON_TXCOMMAND` one?

Atm_nexSerial_RX_TX& Atm_nexSerial_RX_TX::onRxcommand( atm_cb_push_t callback, int idx ) {
  // onPush( connectors, ON_RXCOMMAND, 0, 1, 1, callback, idx );
  onrxcommand.set( callback, idx);
  return *this;
}
Atm_nexSerial_RX_TX& Atm_nexSerial_RX_TX::onTxcommand( atm_cb_push_t callback, int idx ) {
  onPush( connectors, ON_TXCOMMAND, 0, 1, 1, callback, idx );

> STRING RETURN FUNCTION HERE?

  return *this;
}

The state table looks like:
```
const static state_t state_table[] PROGMEM = {
    /*                      ON_ENTER  ON_LOOP  ON_EXIT  EVT_STREAM_START  EVT_START_TXT  EVT_PROCESS_TXT  EVT_START_INT  EVT_PROCESS_INT  EVT_START_CMD  EVT_PROCESS_CMD  EVT_CLOSE_CHARS  EVT_STREAM_CLOSE         ELSE */
    /*  CONNECTION */ ENT_CONNECTION,      -1,      -1,         START_RX,            -1,              -1,            -1,              -1,            -1,              -1,              -1,               -1,          -1,
    /* TX_RECEIVER */             -1,      -1,      -1,               -1,            -1,              -1,            -1,              -1,            -1,              -1,              -1,               -1,  CONNECTION,
    /*    SEND_SET */   ENT_SEND_SET,      -1,      -1,               -1,            -1,              -1,            -1,              -1,            -1,              -1,              -1,               -1, TX_RECEIVER,
    /*    SEND_GET */   ENT_SEND_GET,      -1,      -1,               -1,            -1,              -1,            -1,              -1,            -1,              -1,        START_RX,               -1,          -1,
    /*    START_RX */             -1,      -1,      -1,               -1,       GET_TXT,              -1,       GET_INT,              -1,       GET_CMD,              -1,              -1,               -1,          -1,
    /*     GET_TXT */             -1,      -1,      -1,               -1,            -1,         FIN_TXT,            -1,              -1,            -1,              -1,              -1,               -1,          -1,
    /*     GET_INT */             -1,      -1,      -1,               -1,            -1,              -1,            -1,         FIN_INT,            -1,              -1,              -1,               -1,          -1,
    /*     GET_CMD */             -1,      -1,      -1,               -1,            -1,              -1,            -1,              -1,            -1,         FIN_CMD,              -1,               -1,          -1,
    /*     FIN_TXT */             -1,      -1,      -1,               -1,            -1,              -1,            -1,              -1,            -1,              -1,              -1,          CONFIRM,          -1,
    /*     FIN_INT */             -1,      -1,      -1,               -1,            -1,              -1,            -1,              -1,            -1,              -1,              -1,          CONFIRM,          -1,
    /*     FIN_CMD */             -1,      -1,      -1,               -1,            -1,              -1,            -1,              -1,            -1,              -1,              -1,          CONFIRM,          -1,
    /*     CONFIRM */             -1,      -1,      -1,               -1,            -1,              -1,            -1,              -1,            -1,              -1,              -1,               -1,  CONNECTION,
  };
```

Would you mind giving me a little more direction in how to structure/enable this?

I would expect to use some kind of parameter in one of the cpp functions to "hand over" the string requesting for the number object on the screen. Would that be this: `Atm_nexSerial_RX_TX& Atm_nexSerial_RX_TX::onTxcommand`? Still haven't figured out how to actually deal with that... 

Thanks for any input around this, helping me further...
Happy X-mas days too.

Tsjompie 

The cpp looks as follows:

```
#include "Atm_nexSerial_RX_TX.hpp"
#include <Atm_nexSerial.hpp>
#include <nexData.h>

nexData nexData;

[... 331 lines cut ...]
  automaton.run();
}

void nexSerialStream_callback( int idx, int v, int up ) {
    // Serial.print(v);
    // Serial.println("\t Switch ID after screen press");
    switch ( v ) {
        case 0:       //Load page 0 Disclaimer
        Serial.println("P0 Disclaimer page");
        break;

        case 1:       //Load page 1 (btn page)
        Serial.println("P1 BTn page");
        break;

        case 2:       //Load page 1 (btn page)
        Serial.println("P2 Matrix page");
        break;

        case 3:       //Load page 3 (SETTINGS page)
        Serial.println("P3 SETTINGS page");
        break;
    }
} 
```