tinkerspy/Automaton #74: Pass value to SM using internal machine communication

Tsjompie, 2020-04-14:
Dear Automaton enthusiast,

Not sure whether I should raise this as an issue, as this is more a question I guess....
But still, I hope someone can clear this for me. 

I build an Automaton SM to deal with incoming and outgoing data via a serial stream.
1. The incoming part is covered using the command machine sample. Based on the incoming chars, triggering a callback using a `const char cmdlist[]`works. 
2. For the outgoing traffic I'd like (need?) to pass a string to the State Machine. When not empty the SM to send the string through the serial connection and move back to the idle state.

What I cannot seem to figure out is how to pass that string (being a serial command to the external device) using State Machine Communication. 
I think I need something like: `led.trigger( led.EVT_ON );` but than with the option to also pass a variable (in this case being the string to send through the serial connection).

I might (again) completely misunderstand this, so any help pushing me forward is appreciated.

Thank you, stay safe

tinkerspy, 2020-04-21:
Keep it simple and just send the string directly to the serial port:

Serial.print( s );

Otherwise create a SM::send() method and send it out that way.

Rgdz,
Tinkerspy

Tsjompie, 2020-05-12:
Got it running, thanks!

closed 2020-05-12 by Tsjompie