tinkerspy/Automaton #75: Can't find atm_mpu6050. Machine

rahkshar, 2020-04-18:
Hi please share Atm_mpu6060 machine defination in multicopter thanks

rawtaz, 2020-04-18:
@rahkshar Can you clarify - are you suggesting that such a machine is referenced somewhere but you cannot find it in that place, or are you requesting that someone do the work to write one?

rahkshar, 2020-04-18:
Hi

Please refer
Automaton/multicopter/examples/mpu6050/mpu6050.ino

There is a declaration as below

// WARNING: quality wiring is critical
enum{ YAW, PITCH, ROLL };
Atm_timer timer;
Atm_mpu6050 mpu;

rawtaz, 2020-04-18:
Please use proper linking when you reference files.

I'm not seeing a repository and file that matches the path you wrote above.

Do you mean [tinkerspy/Automaton-Multicopter/examples/mc_mpu6050/mc_mpu6050.ino](https://github.com/tinkerspy/Automaton-Multicopter/blob/master/examples/mc_mpu6050/mc_mpu6050.ino)?

rahkshar, 2020-04-18:
Yes that's correct,

 sorry about bad path reference by me .

rahkshar, 2020-04-19:
Hi

Just wanted to know if the issue raised by me is correct or not ?

The link
Automaton-Multicopter <https://github.com/tinkerspy/Automaton-Multicopter>/
examples
<https://github.com/tinkerspy/Automaton-Multicopter/tree/master/examples>/
mc_mpu6050
<https://github.com/tinkerspy/Automaton-Multicopter/tree/master/examples/mc_mpu6050>
/*mc_mpu6050.ino*
Atm_mpu6050 mpu; is not defined as a machine

rawtaz, 2020-04-19:
I tried to find that Atm_mpu6050 too and failed. I'm not sure where it's supposed to be, and I'm puzzled as to how that example is expected to work :) I'm curious what @tinkerspy will say :)

rahkshar, 2020-04-19:
Ha ha ha ha

Well tinkerspy is awesome person I love his codes hope he will help : )

tinkerspy, 2020-04-19:
The Multicopter code is just not finished. I never got my DIY drone stable (not even running other people's working code) and at some stage Dell formatted my laptop drive so I lost a lot of code. Then I abandoned the project  and started coding a pinball machine ;-)

Sorry to disappoint.

(the pinball machine works fine, though)

tinkerspy, 2020-04-19:
The issue is probably that I made two versions for the mpu6050 module based on two different code sources. They can be found here:

https://github.com/tinkerspy/Automaton-Multicopter/tree/master/src/imu

The example is probably from when I just used a single module, I later split them up.

rahkshar, 2020-04-19:
Thank you Tinkerspy.

If possible can you share atm_mpu6050 machine code please

rawtaz, 2020-04-19:
> Then I abandoned the project and started coding a pinball machine ;-)

Can we see that pinball machine anywhere, I mean the "real" one (physical)?

tinkerspy, 2020-04-19:
Look in the older commits, like this one:

https://github.com/tinkerspy/Automaton-Multicopter/tree/55ca8eaa0ae96629979177c1c2ea76ad7f089885/src

Regards,
Peter van der Landen

tinkerspy, 2020-04-19:
> > Then I abandoned the project and started coding a pinball machine ;-)
> 
> Can we see that pinball machine anywhere, I mean the "real" one (physical)?

You can see the code in my repo, the physical machine I rebuilt is a 1974 Williams OXO, I made it completely event driven.

[image]

The solenoids are driven via optical MOSFET switches that are controlled with addressable neopixel leds. You can see them in operation in this short under-the-playfield video:

https://www.youtube.com/watch?v=Zw2-rP9rN_Y

The idea is that components are coded in an application specific bytecode language, the following code describes a standard bumper (std_bumper):

~~~
init
L, -1, -1, arg_coil     
L, -1, -1, arg_led;      
press
P, -1,  0, -1
H, -1, -1, arg_coil     
T, -1, -1, out_score
J, arg_led, 0, 2     
T, -1, -1, out_score_lit
J, -1, -1, -1
T, -1, -1, out_score_unlit;
turn_on
H, -1, -1, arg_led
T, -1, -1, out_light_on;
turn_off
L, -1, -1, arg_led
T, -1, -1, out_light_off;
~~~

The configuration links it all together, this fragment creates the 'C' bumper and links it to the events it triggers:

~~~
device bumper_c led_bumper_c_grp std_bumper  
link bumper_c out_score_unlit counter0 pt100   
link bumper_c out_light_on save_gate on 
link bumper_c out_light_off save_gate off 
~~~

The idea is that you can define an entire machine at runtime, not at compile-time.

The machine is fully functional now, although I'm still playing with the idea to replace the bytecode thing with a basic based language. It would be cool to run a pinball machine on an event driven multitasking basic engine...

rahkshar, 2020-04-20:
Thank you Peter.

rahkshar, 2020-04-20:
Awesome !!