tinkerspy/Automaton #77: Can't be compiled on ESP32

matthew798, 2020-05-30:
Since the ESP32's arduino libraries don't support some functions like `analogWrite`, compilation fails because some of the `Atm`s use it. I was hoping there would be a workaround other than editing the library's source files manually as it will just be restored by PIO.

delmosaurio, 2020-09-12:
I'm have the same problem. I finded some posible solutions, but i not fully tested this librar.

https://github.com/lbernstone/Tone32
https://github.com/ERROPiX/ESP32_AnalogWrite

gmkado, 2024-02-28:
Not sure if this is relevant, but I'm able to build fine on my esp32 board using platformio with platform = [espressif32](https://registry.platformio.org/platforms/platformio/espressif32)