tinkerspy/Automaton #84: how to use framework with AJAX/websockets

ArnoldGoat, 2021-09-12:
I was struggling with removing 'delay' and substituting with a simple FSM, but it quickly got out of hand once I got beyond a simple demo, so your framework is absolutely great for parallel activities - thanks! I have an existing simple ESP8266WebServer-based website showing some readings. It uses AJAX to update sensor data to avoid whole page refresh. Do you have any suggestions about where to start this sort of thing using your framework? TIA @tinkerspy

tinkerspy, 2021-09-14:
I think I would just create a web server that spits out json and reload that with javascript/ajax. It's not the most efficient way, but for a low traffic device it should be OK.

Regardz,
Tinkerspy

ArnoldGoat, 2021-09-14:
Thanks. I just wanted to head off in a sensible direction. 

Ken Maynard, 83 Canterbury Street, Lyttelton, NZ 8082 ***@***.***) Tel: (03) 328 9553 Mobile: 021 645 894