tinkerspy/Automaton #89: Machine template editor website is unreachable

FarFlyer, 2022-08-05:
Currently, [Machine template editor](http://wolkendek.nl/atm/) is unreachable.

tinkerspy, 2022-08-05:
Yeah. I moved to a tropical island, and while I took a miniaturized version of my server with me, my provider doesn't allow me to open up incoming ports. Unless someone else is able to host it (code is on github), the editor is down for the moment.

Rgdz, Tinkerspy

FarFlyer, 2022-08-05:
Thanks for the quick response.

I hope you're enjoying the tropical island.

Best regards,

Davide Tagliafierro
Mechatronics Engineer
www.daveamps.it
Mobile: +39 329 067 0099

[from mobile]

dsteinfeld, 2024-01-14:
I've brought up the machine editor (at least for now) at: https://machine-editor.thesteinfelds.org.  All are welcome to use it!  A little payback for this terrific library.

dsteinfeld, 2024-01-15:
@tinkerspy feel free to point https://wolkendek.nl/atm/ to https://machine-editor.thesteinfelds.org

tinkerspy, 2024-01-15:
I pointed the wiki URL to your installation.

Thanks a lot!

BlackJackMC, 2024-09-14:
[https://machine-editor.thesteinfelds.org/](https://machine-editor.thesteinfelds.org/) is now unreachable