tinkerspy/Automaton #92: Sleep, what's it good for?

sjernigan, 2024-02-08:
Can you explain the purpose of the sleep flag on a state?

tinkerspy, 2024-02-08:
It's in the Machine class documentation:

https://github.com/tinkerspy/Automaton/wiki/Automaton-Machine-class#uint8_t-sleep-int8_t-v-

> 
A machine in a sleeping state does not execute its event loop and does not call its action() handler, it does, however, process incoming messages.

sjernigan, 2024-02-09:
Thank you.  By "incoming messages", do you mean just custom methods?  Does a machine put to sleep delay any state timers.  Would they not occur if the machine were asleep when it expired (or maybe it occurs immediately on wake up) or it expires in the future as if the machine were never asleep?